dotintent/react-native-ble-plx #489: Device was Disconnected - monitorCharacteristicForService

ExoMemphiz, 2019-06-22:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Monitoring a characteristic for notifications, should receive the notifications.

## Current Behavior

Receive an error that is not very detailed, I tried adding logs to the java code, to see what happens and where. This is as far as I've come:

In BleModule.java, the function "safeMonitorCharacteristicForDevice", the setupMode is set to "QUICK_SETUP", and it setup a notification (
```java
if ((properties & BluetoothGattCharacteristic.PROPERTY_NOTIFY) != 0) {
                    Log.e("BleModule", "[safeMonitorCharacteristicForDevice] setupNotification");
                    return connection.setupNotification(gattCharacteristic, setupMode);
}
``` 

And the error function gets invoked:
```java
@Override
public void onError(Throwable e) {
                        Log.e("BleModule", "[safeMonitorCharacteristicForDevice] onError error toString: " + e.toString());
                        Log.e("BleModule", "[safeMonitorCharacteristicForDevice] onError error message: " + e.getMessage());
                        errorConverter.toError(e).reject(promise);
                        transactions.removeSubscription(transactionId);
}
```

The Log.e produce the following:

```
E/BleModule: [safeMonitorCharacteristicForDevice] setup mode: Quick Setup
E/BleModule: [safeMonitorCharacteristicForDevice] setupNotification
E/BleModule: [safeMonitorCharacteristicForDevice] onError error toString: com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from .... with status 0 (GATT_SUCCESS)
E/BleModule: [safeMonitorCharacteristicForDevice] onError error message: Disconnected from .... with status 0 (GATT_SUCCESS)
```
Where "...." is a unique device ID.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. writeCharacteristicWithoutResponseForService for the RECORD_ACCESS_CONTROL_POINT, with the byte values [1, 1], encoded in Base64 format.
2. monitorCharacteristicForService for the Glucose Measurement (00002A18) characteristic.
3. Experience the error.

### Context

* Library version: "react-native-ble-plx": "^1.0.3"
* Platform: Android

dariuszseweryn, 2019-06-24:
Could you add more context?
The minimal code that shows the issue?
So far it seems that you have a problem with the peripheral you use

ExoMemphiz, 2019-06-24:
Sure, we can assume I am connecting correctly, as I can read "normal" values, such as its serial number just fine.

Here is a snippet of the code to set up the "monitoring" of the 2 Glucose Characteristics, "Measure" and the "Measure Context"

https://hastebin.com/nefuvusaro.js

ExoMemphiz, 2019-06-24:
Also, just tried running the code again, got this, this time:

```
{"message":"Device .... was disconnected","errorCode":201,"attErrorCode":19,"iosErrorCode":null,"androidErrorCode":null,"reason":"Disconnected from .... with status 19 (UNKNOWN)"}
```

dariuszseweryn, 2019-06-24:
`"attErrorCode":19` is `GATT_CONN_TERMINATE_PEER_USER`. The peripheral you use disconnects itself. Most probably because the bonding is not established. Bonding is managed by the system and the library does nothing with it.

You should ask the firmware developer what is the preferred way of interacting with this device and stick to it.

ExoMemphiz, 2019-06-24:
Fantastic, have contacted them, awaiting their response.
If it's somewhat useable for others, and not ultra specific for my use case I will leave a comment with the answer that I got, otherwise I will close this issue, thanks for your swift replies!

closed 2019-08-30 by stale[bot]

tymoxx, 2021-07-14:
@ExoMemphiz 
> Fantastic, have contacted them, awaiting their response.

Did you find out what was the reason for that error?

joaovitorqueiroz, 2021-07-17:
I'm also having the same problem, @tymoxx you are also having this problem?

tymoxx, 2021-07-18:
@joaovitorqueiroz In my case, the device was disconnecting after 45 seconds. 
So I added `setInterval` to `writeCharacteristicWithoutResponseForService` (which just gets status about the device) for every 40seconds. And it keeps the device connected.

harguus, 2021-11-12:
> @joaovitorqueiroz In my case, the device was disconnecting after 45 seconds. So I added `setInterval` to do `writeCharacteristicWithoutResponseForService` every 40 seconds(which just gets the status of the device). And it keeps the device connected.

Hello, I have the same problem and wanted to implement your solution, where I put the setInterval, could you show me the code.

Thanks!

mhb-yt, 2021-11-12:
@harguus I put `setInterval` inside the `connect()` function. 
Here is a simplified code. Your code will be different depending on how you connect to your device:

```javascript

async connect() {
            const device = await deviceManager.connectToDevice(id);
            const discoveredDevice = await device.discoverAllServicesAndCharacteristics();

/* Sync periodically - to keep device connected*/
            const interval = setInterval(function () {
                const value = 'your_value'; // in my case, I synchronize time between the app and device. But you can do anything - it's just to keep the device connected.
                discoveredDevice.writeCharacteristicWithResponseForService(SERVICE_UUID, WRITE_CHARACTERISTIC_UUID, value);
            }, 40000);
            dispatch(saveInterval(interval)); // I also save the instance of the interval to the state, so I can do clearInterval later.

}
```

tymoxx, 2021-11-12:
@harguus I put `setInterval` inside the `connect()` function. 
Here is a simplified code. Your code will be different depending on how you connect to your device:

```javascript

async connect(id) {
            const device = await deviceManager.connectToDevice(id);
            const discoveredDevice = await device.discoverAllServicesAndCharacteristics();

            /* Sync periodically - to keep device connected*/
            const interval = setInterval(function () {
                const value = 'your_value'; // in my case, I synchronize time between the app and device. But you can do anything - it's just to keep the device connected.
                discoveredDevice.writeCharacteristicWithResponseForService(SERVICE_UUID, WRITE_CHARACTERISTIC_UUID, value);
            }, 40000);
            dispatch(saveInterval(interval)); // I also save the instance of the interval to the state, so I can do clearInterval later.
}
```

harguus, 2021-11-12:
> @harguus I put `setInterval` inside the `connect()` function. Here is a simplified code. Your code will be different depending on how you connect to your device:
> 
> ```js
> async connect(id) {
>             const device = await deviceManager.connectToDevice(id);
>             const discoveredDevice = await device.discoverAllServicesAndCharacteristics();
> 
>             /* Sync periodically - to keep device connected*/
>             const interval = setInterval(function () {
>                 const value = 'your_value'; // in my case, I synchronize time between the app and device. But you can do anything - it's just to keep the device connected.
>                 discoveredDevice.writeCharacteristicWithResponseForService(SERVICE_UUID, WRITE_CHARACTERISTIC_UUID, value);
>             }, 40000);
>             dispatch(saveInterval(interval)); // I also save the instance of the interval to the state, so I can do clearInterval later.
> }
> ```

ok i got it, sorry for my amateur question but i don't know what i send in value. In the documentation it says to send a base64 value. what would that value be?

harguus, 2021-11-13:
> @harguus I put `setInterval` inside the `connect()` function. Here is a simplified code. Your code will be different depending on how you connect to your device:
>
> ```js
> async connect(id) [...]

Hello, thanks for the light at the end of the tunnel, I understood well, I used setInterval() but calling the write function passing a command to return the device version. resolved well so far.

Thank you!