dotintent/react-native-ble-plx #490: [iOS] Unable to read battery percentage

phil3903, 2019-07-02:
Quick recap:
-iPhoneX, iOS 12.3.1, react-native-ble-plx 1.0.3
- I am able to read services and characteristics after calling 'discoverAllServicesAndCharacteristic' and continuing down the promise chain
- I've attempted this with 3 different devices. They all return various services and characteristics values (firmware versions, manufacturer name, etc.) but none of them are battery level services or characteristics.
- I'm new to interacting with Bluetooth, but after reading this: https://stackoverflow.com/questions/19539535/how-to-get-the-battery-level-after-connect-to-the-ble-device/19701412 and other various posts, I think I've come to realize the UUIDs for battery service and characteristic are standardized via the bluetooth spec?
- After attempting to access my devices battery level with these UUIDs via 'servicesForDevice' and 'characteristicsForDevice',  I get an error back that saying neither exists. 
- I checked out this repos index.d.ts, and saw you had an example setup for monitoring. I attempted that implementation as well, but no luck.

Hopefully I'm not doing something stupid, but any help would be greatly appreciated.

closed 2019-07-23 by phil3903

OisinOKeeffe, 2020-01-10:
@phil3903  did you end up getting this working?

d0xi5, 2020-09-17:
Nothing on this?

Thanks