dotintent/react-native-ble-plx #492: `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod

wgod58, 2019-07-05:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [Y] I am running the latest version
- [Y] I checked the documentation and found no answer
- [Y] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?
I follow this [https://blog.expo.io/so-you-want-to-build-a-bluetooth-app-with-react-native-and-expo-6ea6a31a151d](url) article to make a project

### Steps to Reproduce
1. step 1

> At step **pod install** came out this error
> [!] Unable to determine Swift version for the following pods:
> 
> - `react-native-ble-plx-swift` does not specify a Swift version and none of the targets (`myapp2`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.

### Solve 
According to this issue [https://github.com/CocoaPods/CocoaPods/issues/7327](url)
I modified react-native-ble-plx-swift.podspec one line code
s.pod_target_xcconfig = { 'SWIFT_VERSION' => '4.1' }  --> s.swift_version = "4.1"
And it works for me

### Here's the code
`
require 'json'

package = JSON.parse(File.read(File.join(__dir__, 'package.json')))

Pod::Spec.new do |s|
  s.name         = "react-native-ble-plx-swift"
  s.version      = package['version']
  s.summary      = "React Native Bluetooth Low Energy library - Swift static library"

  s.authors      = { "Przemysław Lenart" => "przemek.lenart@gmail.com" }
  s.homepage     = "https://github.com/Polidea/react-native-ble-plx#readme"
  s.license      = "Apache License 2.0"
  s.platform     = :ios, "8.0"

  s.source       = { :git => "https://github.com/Polidea/react-native-ble-plx.git" }
  s.source_files  = "ios/**/*.{swift}"
  s.swift_version = "4.1"
end
`

### Context
* Library version:  1.0.3
* Platform: Android/iOS
* Platform logs (logcat/XCode):10.2,1

jono-allen, 2019-07-07:
what react-native version are you using?

wgod58, 2019-07-08:
Here's my package.json
`{
  "main": "node_modules/expo/AppEntry.js",
  "scripts": {
    "start": "expo start",
    "android": "expo start --android",
    "ios": "expo start --ios",
    "web": "expo start --web",
    "eject": "expo eject"
  },
  "dependencies": {
    "exp": "^57.2.1",
    "expo": "^33.0.0",
    "expokit": "33.0.4",
    "react": "16.8.3",
    "react-dom": "^16.8.6",
    "react-native": "https://github.com/expo/react-native/archive/sdk-33.0.0.tar.gz",
    "react-native-ble-plx": "^1.0.3",
    "react-native-unimodules": "^0.4.0",
    "react-native-web": "^0.11.4"
  },
  "devDependencies": {
    "babel-preset-expo": "^5.1.1"
  },
  "private": true
}
`

shunitoh, 2019-07-10:
Open XCODE and add SWIFT_VERSION to User-Defined in the project settings, it worked.
[image]

hugAdmin, 2019-07-22:
Just to bump this up. I am also having the exact issue.

hugAdmin, 2019-07-23:
I discovered this thread https://github.com/kiwi-bop/flutter_crashlytics/issues/22 and the solution of adding a new Swift file in the project fixed the issue for me

rezasazesh, 2019-09-23:
I solved this issue by making sure the pod is NOT under `target '{PROJECT}-tvOSTests' do` section once I moved the pods in the main pod section it was resolved. When you link by default the pods are placed under the tvOSTests section for some reaston
````
  pod 'react-native-ble-plx', :path => '../node_modules/react-native-ble-plx'
  pod 'react-native-ble-plx-swift', :path => '../node_modules/react-native-ble-plx'

````

Baskerville42, 2019-11-20:
Add new swift file and select the correct target and this issue will be solved

wgod58, 2019-12-06:
look like there is so many solutions
it has not had recent activity. It will be closed if no further activity occurs.

closed 2019-12-06 by wgod58

ccnasro, 2023-05-18:
Try to add the swift version in the top of your Podfile
`ENV['SWIFT_VERSION'] = '3'` for example