dotintent/react-native-ble-plx #493: Android install instructions are incomplete.

Xesme, 2019-07-09:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
(This is an ongoing issue within your issues, updating the install instructions would resolve.) 

## Current Behavior

What is the current behavior?
Currently the install instructions for Android are incomplete. The MainApplication.java should be updated to reflect changes for soLoader, BLEPackage, and MainReactPackage. If you follow the current instructions you will run into `TypeError: undefined is not an object (evaluating '_BleModule.BleModule.createClient') ` when running a build on a device. 

## New behavior

Please describe the behavior you are expecting.

I am expecting to be able to install the lib from the README.md instructions. 

### Acceptance criteria

Please provide detailed steps to accomplish proposed feature.

In MainApplication.java add: 
```
import com.polidea.reactnativeble.BlePackage;
import com.facebook.react.shell.MainReactPackage;
import com.facebook.soloader.SoLoader;
```

```
public class MainApplication extends Application implements ReactApplication { ... 
protected List<ReactPackage> getPackages() {
    return Arrays.<ReactPackage>asList(
           new MainReactPackage(),
           new BlePackage()
... 
```
AND
```
  @Override
  public void onCreate() {
  super.onCreate();
  SoLoader.init(this, /* native exopackage */ false);
  }
}
``` 

[Here](https://github.com/Cierpliwy/SensorTag/blob/master/android/app/src/main/java/com/sensortag/MainApplication.java) is another example of the necessary steps for MainApplication.java
### Breaking changes

Does a feature breaks compatibility with previous version?

This is an ongoing issue with all versions.

Charanor, 2019-09-11:
This did not fix the issue for me unfortunately. This issue as a whole seems to be a problem with Expo (after ejecting), because I see people saying "This is an issue with Expo but I fixed it" (and then not saying how >_>).

Xesme, 2019-09-11:
@Charanor I don't use expo for React-Native so I wont be much help to you! Ejecting may be your largest problem if you used create react app.

closed 2019-11-17 by stale[bot]