dotintent/react-native-ble-plx #494: Bug Report: [iOS Background & Suspended mode]"the callback never gets invoked in the startDeviceScan "

GeekChao, 2019-07-11:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [Yes ] I am running the latest version
- [Yes ] I checked the documentation and found no answer
- [ Yes] I checked to make sure that this issue has not already been filed

## Expected Behavior
When the app is even in the background or suspended mode of ios, the callback of the startDeviceScan gets invoked.

## Current Behavior
When the app is in the background or suspended mode of ios, the callback of the startDeviceScan never gets invoked. I assume there is never discovery event found yet, according to the apple ble doc. Is it possible to configure the 'startDeviceScan' to make the app scan in the background and suspended mode? Thank you
### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context
IOS Background and suspended mode

* Library version: 1.0.3
* Platform: iOS.
* Platform logs (XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

closed 2019-07-15 by GeekChao

jayrenteria, 2019-09-13:
@GeekChao did you find a solution for this?

joaomdmoura, 2021-06-14:
Same issue, any updates?