dotintent/react-native-ble-plx #498: react-native iOS build with fastlane + jenkins

jgimeneztc, 2019-07-15:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.
Hi, I'm building my project with jenkins, and I'm having a problem with adding file.swift and header without human interaction(step 3 https://github.com/Polidea/react-native-ble-plx#ios-pure-react-native-example-setup). 
If I add the files manually in Xcode the app compile without problem.
 
How can this step be done on jenkins automaticaly?  or how can I solve this?
Thanks

piotrdubiel, 2019-07-16:
If you need to do this automatically, you can take a look at https://github.com/Polidea/react-native-ble-plx/blob/master/integration-tests/scripts/enable-swift. It's a simple ruby script that uses `xcodeproj` gem to add `File.swift` and bridging header.

jgimeneztc, 2019-07-17:
@piotrdubiel thanks

closed 2019-07-17 by jgimeneztc