dotintent/react-native-ble-plx #499: Device.serviceUuids always returns empty

karlmarxlopez, 2019-07-16:
I am using an `ESP32` MCU as a BLE server. When I access `device.serviceUuids` when scanning starts, it always returns `null`. 

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior
It should return array of services UUIDs

Please describe the behavior you are expecting

## Current Behavior
Returns null

### Steps to Reproduce

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: latest
* Platform: Android

Cierpliwy, 2019-08-27:
`device.serviceUuids` is a value obtained from advertisement data, which may be `null`. To check "normal" services you have to call [`servicesForDevice(deviceIdentifier)`](https://polidea.github.io/react-native-ble-plx/#blemanagerservicesfordevice)

closed 2019-08-27 by Cierpliwy

rafael-metractive, 2019-09-17:
@karlmarxlopez 

Name given to ble cannot be longer than 3 characters
Take a look at this. We spent a lot of time here on a project to find out in a forum about this.