dotintent/react-native-ble-plx #501: Android, connectToDevice() repeatedly fails after a few successes

GreanBeetle, 2019-07-17:
**react-native-ble-plx 1.0.3**
**react-native 0.59.1**

**Android-only** issue.

Our app connects to washers and dryers via bluetooth. Our `MachineListScreen` lists available machines

`machineA`
`machineB`
`machineC`

Using `connectToDevice(machine)`, we are successfully able to connect to `machineA`, `machineB`, and `machineC`, transition to `ConnectedToMachineScreen` and run interactions.

This works only once!

The next time we `connectToDevice(machine)`, it simply fails to connect. The error is a `DeviceConnectionFailed`. It's due to a connection attempt timeout that we impose, but this is immaterial. Our timeout for a connection attempt is 10 seconds. But we've tried bumping it to 30 seconds, 60 seconds, et cetera, and it still fails.

This doesn't happen on iOS. iOS connects repeatedly.

**Question**

On Android, is it possible we need to `BluetoothGatt.refresh(`) before re-connecting to a machine? If so, does anyone know how to do this using this `connectToDevice()` option:

[`connectToDevice(machine, { refreshGatt: ? })`](https://polidea.github.io/react-native-ble-plx/#connectionoptions) What is the value for this key?

Failing that, does anyone have a different suggestion?

Here's the code.
 
```js
_handleMachinePress = async machine => { 
  try {  
    this.setState({ loading: true });
    await ble.connectionManager.connectToMachine(machine);
    const passProps = { ...props };
    navigateToScreen(ConnectedToMachineScreen, passProps);
  } catch (error) {
    console.warn(error);
  } finally {
    this.setState({ loading: false });
  }
}
```

```js
export const connectToMachine = async machine => {  
  return new Promise(async (resolve, reject) => {
    let connectedDevice;
    try {
      connectedDevice = await ble.connectToDevice(machine.id, { timeout: 10000, autoConnect: true });
    } catch (error) {
      try {
        await timeout(5000);
        connectedDevice = await ble.connectToDevice(machine.id, { timeout: 10000 });
      } catch (error) {
        reject(error);
        return;
      }
    }
    try {
      const exploredDevice = await connectedDevice.discoverAllServicesAndCharacteristics();
      await exploredDevice.services();
      resolve();
    } catch (error) {
      reject(error);
    }
  });
};
```

folofse, 2019-07-19:
Experiencing the same problem, after a disconnect the app fails to reconnect to the peripheral, this occured after upgrading to React Native 0.60.x. I get isDeviceConnected false, but when I try to reconnect i get the error 203 Device is already connected.

GreanBeetle, 2019-07-22:
@folofse So, I found a hacky solution to this. I don't love it. But this has worked for this issue

```js
let ble = new BleManager()

ble.startDeviceScan()

ble.destroy()

ble = new BleManager()

ble.connectToDevice()
```

Again, I think this is a poor solution. If someone has a better alternative, please advise.

folofse, 2019-07-22:
@GreanBeetle Yes it seems like a new instance of BLEManager will solve the problem. I should dig around in the code a bit and see what might be the root of the problem between connects. 

I also had a similar problem with isDeviceConnected=false with the other BlueTooth package https://github.com/innoveit/react-native-ble-manager so it might be a Android related problem that it returns isDeviceConnected=false even if its currently connected.

GreanBeetle, 2019-07-24:
Still an issue. @Cierpliwy Any thoughts?

kyoyadmoon, 2019-07-25:
same issue here, I can repeat `connect` and `cancelConnection` once or several times. But after a few success, `connect` function will keep execute without any response. I added timeout option to `connect` and get the timeout Error: `ERROR: Operation was cancelled`. The error repeat until I reopen the app.

it could be some bug with `cancelConnection`. 
My work around is do disable and immediately re-enable bluetooth to cancel connection instead of call `cancelConnection`.

hope that helped.

GreanBeetle, 2019-07-26:
@kyoyadmoon I'll give that a try

chaselal, 2019-08-01:
I've seen a problem that might be related. If I call BleManager.connectToDevice and then call BleManager.cancelDeviceConnection before the device connects, then the devices becomes connected and the BleManager instance reports that the device is not connected. Calling BleManager.cancelDeviceConnection does not release the connection. Even closing my app does not release the connection. Calling BleManager.disable releases the connection. I assume BleManager.destroy would release the connection, but I have not tried it.

GreanBeetle, 2019-08-01:
@chaselal In our case, we run somewhat complex transactions with firmware. There are numerous `writeCharacteristicWithResponseForDevice()` calls before we `cancelDeviceConnection`. When we _do_ cancel, what I've discovered is that `react-native-ble-plx` just breaks. All methods stop working: `connectToDevice()` fails, `startDeviceScan()` fails, `destroy()` fails ... 

I've resorted to a crappy work-around using `react-native-restart` to restart the app and wipe out the `BleManager` ...

chaselal, 2019-08-01:
@GreanBeetle I doubt this affects your problem, but there is a bug in your code. At least, it looks like one. `connectToMachine = async machine ...` should be `connectToMachine = machine ...` because currently `connectToMachine` returns a `Promise<Promise>`. You probably want it to just return a `Promise`. The extra `async` introduces the extra Promise. What happens now is awaiting `connectToMachine` resolves immediately, instead of resolving after the machine connects.

GreanBeetle, 2019-08-01:
@chaselal Ah yeah we `await` that

chaselal, 2019-08-01:
Do you `await await connectToMachine(machine)`? You need two awaits for the
code you posted.

GreanBeetle, 2019-08-01:
Nope. Just the one. 

```js
import { connectToMachine } from 'utils'

connect = async () => { await connectToMachine(m) }
```

glenne, 2019-08-24:
I see the same issue here.  I believe the root problem is that cancelDeviceConnection does not end up calling 'disconnect' (or close in some cases) on the underlying BluetoothGatt instance.  It's leaving all the underlying android state in limbo.

GreanBeetle, 2019-09-03:
Our fix ultimately looked something like this

```js
## BLEManager/index.js ##
import { BleManager } from 'react-native-ble-plx';

export const create = () => { 
  ble = new BleManager();
};

export const setLogLevel = logLevel => {
  ble.setLogLevel(logLevel);
};

export const restart = () => { 
  ble.destroy(); 
  create();
};

export const cancelConnection = machineId => ble.cancelDeviceConnection(machineId);

create();

export let ble;
```

After a successful interaction with a (slave) device, we `cancelConnection()` and `restart()`. We went through numerous iterations and, for whatever reason, this particular setup worked.

WilliamAlexander, 2019-09-08:
@GreanBeetle I too am seeing Operation was cancelled. Seems random for that moment, I tried your suggestion. Seems more reliable after a little testing. Still not perfect though

ThomasStubbe, 2019-09-19:
We're having the same issue. After calling `cancelDeviceConnection` the BleManager is stuck, no response on any method.

closed 2019-11-25 by stale[bot]

glenne, 2019-11-25:
Sad this is being retired when it's still an issue with no reasonable workaround.  If I have more than one device I have to abort all and reconnect just to disconnect one.

ScreamZ, 2019-12-20:
ping @dariuszseweryn 

Experiencing the same issue:

```js
    console.log(await to(bleManager.cancelDeviceConnection(deviceId)));
    const [errConnect, device] = await to(bleManager.connectToDevice(deviceId));
    console.log(errConnect);
```
[image]

It occurs when I disable bluetooth while being connected…

GreanBeetle, 2019-12-20:
Yes, this is still a common issue for us. 

```js
let ble = new BleManager()
console.log("Manager 1", ble)
ble.destroy()
ble = new BleManager()
console.log("Manager 2", ble)
```
After this, any `ble` method we call (e.g. `ble.state()`) freezes our app. 

Console logging the ble manager instance, I've found one difference 

(a) A working ble manager looks this

[image]

(b) A faulty ble manager that will freeze the app looks like this. Note that it has a single unresolved promise in the `_activePromises` property 

[image]

dariuszseweryn, 2019-12-20:
Please add related info according to [the issue template](https://github.com/Polidea/react-native-ble-plx/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) with special care of adding logs and native logs as described.

gregogalante, 2020-07-31:
@GreanBeetle I get the same problem with pending promises. Have you found a solution?

alvinycheung, 2022-03-15:
I'm seeing this problem with both iOS and Android when trying to connect to a device. Connections don't seem to be working at all right now.