dotintent/react-native-ble-plx #507: updated iPhone to iOS 12.4, BLE scan() stopped working

ravelab, 2019-07-23:
## Expected Behavior

scan() return BLE devices

## Current Behavior

scan() doesn't return any devices

### Steps to Reproduce

1. update iOS to 12.4
2. try BLE app again

### Context

this is what I see from log in XCode:

2019-07-23 14:53:10.206294-0700 appble[1665:399596] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x28360cbc0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2019-07-23 14:53:10.208288-0700 appble[1665:399617] [] nw_socket_handle_socket_event [C5:1] Socket SO_ERROR [61: Connection refused]
2019-07-23 14:53:10.211523-0700 appble[1665:399601] [] nw_connection_get_connected_socket [C5] Client called nw_connection_get_connected_socket on unconnected nw_connection
2019-07-23 14:53:10.211588-0700 appble[1665:399601] TCP Conn 0x281600000 Failed : error 0:61 [61]

ravelab, 2019-08-02:
started working after adjusting my code, sorry for the false alarm

closed 2019-08-02 by ravelab

cdeskins, 2019-08-02:
@ravelab What did you have to change?  Seeing similar issues in our implementation of the library.

agentmusic, 2019-08-23:
@ravelab @cdeskins What did you have to change? In my app scan() works but less reliably than in previous iOS versions (versions before iOS 12.4)

caominhthe, 2019-09-17:
@ravelab what did you do sir? i got the same problem on ios 12.4

ravelab, 2019-09-19:
@cdeskins @agentmusic @caominhthe sorry for the late reply. Scan() was actually always working for me, I just needed to filter the list of devices returned correctly by localName