dotintent/react-native-ble-plx #508: Connection to two devices and recover services

keyz23, 2019-07-24:
Hi everyone, I'm using this library to connect to two devices that should be used simultaneously.

I don't know how to do, so I'm using the connection to first device (right device) and then the connection to second one (left device).

After the connection (that I think has been done) I have to be able to recover the services and features (all this I was able to do for a single device).

With two I don't know how to implement it. At the moment I tried to connect only to a dispostivo but I get a:

> Possible Unhandled Promise Rejection (id: 0):
> TypeError: Cannot read property 'monitorCharacteristicForService' of undefined
> TypeError: Cannot read property 'monitorCharacteristicForService' of undefined

how can I recover the services and characteristics simultaneously from the two devices? thank you.

```
scanDx() {
    var flag_dx = false;
    this.manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        return
      }
      console.log("this.model_dx: " + this.model_dx(this.props.ShoesModel));
      console.log("Device.Name: " + device.name)
      var device_dx = device.name
    if (device_dx == this.model_dx(this.props.Model)) { 
      console.log("device_dx: " + device_dx)
      this.manager.stopDeviceScan();
      device.connect()
        .then( () => {
          console.log("CONNESSO DX")
          flag_dx = true;
          console.log("DX flag: " + flag_dx)
          this.scanSx({device_dx: device_dx });
        })
        .catch(() => {
          Alert.alert("No Connection.")
          Actions.homepage();
        });
    }
  });
}

  scanSx({device_dx: device_dx }) {
  var flag_sx = false;
    this.manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        return
      }
    console.log("this.model_sx: " + this.model_sx(this.props.Model));
    console.log("Device.Name: " + device.name)
    var device_sx = device.name
    if (device_sx == this.model_sx(this.props.Model)){
      console.log("device_sx: " + device_sx)
      this.manager.stopDeviceScan();
      device.connect()
      .then( () => {
        console.log("CONNESSO SX")
        flag_sx = true;
        console.log("Left Flag: " + flag_sx)
        this.deviceService(device);
      })
      .catch( () => {
        Alert.alert("No Connection.")
        Actions.homepage();
      }); 
  }
});
} 

  deviceService(device) {
 if (device.name == this.model_sx(this.props.ShoesModel)) {
      device.discoverAllServicesAndCharacteristics()
        .then(response => {
          console.log("try1: " + Object.keys(response));
          console.log("serviceUUIDs: " + response.serviceUUIDs);
        })
        .then(response2 => {
          console.log("try2: " + this.setupNotifications(response2));
        })
        .catch(error => {
          this.error(error.message);
        });
    }
  }

  async setupNotifications(device) {
    const service = this.serviceGeneral();
    device.monitorCharacteristicForService(
      service,
      this.AccGyrMg,
      (error, characteristic) => {
        if (error) {
          this.error(error.message);
          return;
        }
 //......
```

keyz23, 2019-08-05:
up

keyz23, 2019-08-27:
Anyone can help me? I would know how to connect to two devices

dariuszseweryn, 2019-08-27:
Well... You do not even try to `device_dx.connect()`. I do not know what you expect?

closed 2019-11-02 by stale[bot]