dotintent/react-native-ble-plx #511: How write date to bluetooth device.

zainlinux1989, 2019-07-24:
1. How write date to bluetooth device
2. Can library provide any guideline to convert different data structure or data into base64

zainlinux1989, 2019-07-25:
>     1. How write date to bluetooth device
> 
>     2. Can library provide any guideline to convert different data structure or data into base64

I find a solution to write date on bluetooth device.

let time = date.getTime();
let seconds = parseInt(time / 1000);
let data = new Buffer(4);
data.writeUInt32BE(seconds);

 this.manager.writeCharacteristicWithResponseForDevice(deviceId, serviceUUID, charUUID, data.toString('base64'))
                .then((success) => {
                    resolve(success);
                })
                .catch((error) => {
                    reject(error)
                })
        });

closed 2019-07-25 by zainlinux1989

fabio-C, 2020-05-13:
@zainlinux1989 
What you used for serviceUUID and charUUID?
Where these variables come from?