dotintent/react-native-ble-plx #512: readCharacteristicForDevice incorrect value read on Android

leonchabbey, 2019-07-26:
## Current Behavior

After base64 decoding, the value returned from readCharacteristicForDevice on an Android device isn't equal to what was sent by the bluetooth device.

### Steps to Reproduce

You have the value: ```aaaaabbbbbccccchhhhhjjjjjkkkkklllllpppppoooooiiiiiuuuuuzzzzzttttt``` which you expect from reading the characteristic.

#### On iOS
```
Characteristic {
  deviceID: "D60CEFD2-3445-C765-ED73-13E2C0EF0D19"
  id: 10789654848
  isIndicatable: false
  isNotifiable: false
  isNotifying: false
  isReadable: true
  isWritableWithResponse: false
  isWritableWithoutResponse: false
  serviceID: 10750760576
  serviceUUID: "5de34279-8679-5cb1-1ef8-f4d9672f508b"
  uuid: "08590f7e-db05-467e-8757-72f6faeb13d5"
  value: "YWFhYWFiYmJiYmNjY2NjaGhoaGhqampqamtra2trbGxsbGxwcHBwcG9vb29vaWlpaWl1dXV1dXp6enp6dHR0dHQ="
}
```

It needs to be (base64) decoded:
```
var value = Buffer.from(valueFromCharacteristic, "base64").toString();
// Great ! The value is what we expected : aaaaabbbbbccccchhhhhjjjjjkkkkklllllpppppoooooiiiiiuuuuuzzzzzttttt
```

#### On Android
```
Characteristic {
  deviceID: "B8:27:EB:2C:91:54"
  id: 8
  isIndicatable: false
  isNotifiable: false
  isNotifying: false
  isReadable: true
  isWritableWithResponse: false
  isWritableWithoutResponse: false
  serviceID: 10
  serviceUUID: "5ceb7awd-4564-5cb1-awda9-f4d9672f508b"
  uuid: "08590f7e-db05-467e-8757-72f6faeb13d5"
  value: "YWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFiYmJiYmNjY2NjaGhoaGhqamFhYWFhYmJiYmJjY2NjY2hoaGhoamphYWFhYWJiYmJiY2NjY2NoaGhoaGpqYWFhYWFi"
}
```
First you can see here the "value" field **is not the same as iOS's**.
base64 decode part:
```
var value = Buffer.from(valueFromCharacteristic, "base64").toString();
// What ? The value is now equal to : aaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaabbbbbccccchhhhhjjaaaaab
```
You can also see here, a small part of the string *aaaaabbbbbccccchhhhhjj * is repeated over and over...
I have tried everything I could think of on the javascript side, so I'm wondering now if this could be a bug in the library. 

### Context

* Library version: 1.0.3
* Platform: Android

dariuszseweryn, 2019-08-01:
Could you add more info as requested by the template?

dariuszseweryn, 2019-08-02:
Specifically I would need to see the native logs

Agupane, 2019-09-02:
I have the same problem. The device sent a '10' value but I got: 
```
Characteristic {
  deviceID: "B8:27:EB:2C:91:54"
  id: 8
  isIndicatable: false
  isNotifiable: false
  isNotifying: false
  isReadable: true
  isWritableWithResponse: false
  isWritableWithoutResponse: false
  serviceID: 10
  serviceUUID: "5ceb7awd-4564-5cb1-awda9-f4d9672f508b"
  uuid: "08590f7e-db05-467e-8757-72f6faeb13d5"
  value: "Vw=="
}
```

closed 2019-11-08 by stale[bot]

HarshaVardhanPVK, 2023-03-24:
I have similar issue getting value as "ABE=" and not able decode it.