dotintent/react-native-ble-plx #514: I need scanRecord from Android

meridPL, 2019-07-31:
Hi,

Some time ago, I wrote an Android app that used the Beacon. To connect a Beacon with an app, I used BluetoothAdapter.LeScanCallback from Android. Now I want to move the android project to the react native project. I researched the internet to find the best solution and I used your solution, but I have a problem with receiving data from the leScanCallback method, where one of the parameters is value of scanRecord of byte type. I need scanRecord , but I don't know how to do it. Let me add that, when I get information from the startDeviceScan method and log data from the device my Beacon doesn't have a UUID, but he has an id.

Kosik-Chris, 2019-07-31:
It sounds as if you are attempting to obtain the raw advertisement data from the packet.
From the documentation you would obtain the device object and then access its static members to look for various advertisement packet types (manufacture data for example 0xFF, 0x07 service UUID etc.).

Complete list: https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/

Here is some basic code I used to obtain mac address and other device object members (very WIP):

`export default class Main extends Component {
  constructor() {
    super();
    this.state = {
      adcValue: null,
      decimalVal: null,
      celciusVal: null,
      farenheitVal: null,
      device: null
    };
    this.manager = new BleManager();
  };

  componentDidMount() {
    const subscription = this.manager.onStateChange((state) => {
      if (state === "PoweredOn") {
        this.scanObserverValues();
        subscription.remove();
      }
      //this.scanObserverValues();
    }, true);
  }

  scanObserverValues(){
    //start device scan 
    //@params = (UUIDs, options, listener)
    //listener function (error, device) obtain device as second param

    //pass in options to increase scan rate/ lower latency = increased device power
    this.manager.startDeviceScan(
      null,
      { options: ScanMode.LowLatency },
      (error, device) => {
      if (error) {
        console.log(error.message);
        return;
      }
      /*
        SW filter implementation: unique field chosen = MAC

        process: initially no MAC is stored/ configured and no data will show up..
        User clicks some button to scan and reveal all devices in vicinity that are 
        broadcast and select (via checkbox) which MAC's we want to save for display
        of values (manufacture data)

        **Also allow feature that allows user to display all raw tx data or
        potentially pick a different field to track (manufacture data, characterisitcs)
        etc.
      */

      //primitive filter..

        if (device.id === "secret_Mac") {
          try {

          this.setState({ device: device });

            console.log(device.id); //MAC
            let rawData = device.manufacturerData;
            //get hex (ascii) values for each character
            //tx packet always only care about first 3 chars ignore = on x64
          let decOne = getDecFrom64(device.manufacturerData.charCodeAt(0));
          let decTwo = getDecFrom64(device.manufacturerData.charCodeAt(1));
            let decThree = getDecFrom64(device.manufacturerData.charCodeAt(2));

          let decVal = [(decOne*262144)+(decTwo*4096)+(decThree*64)]/256;
            //let decVal = [819456/256];

          /*where 3200 = 25.00 deg celcius
          resolution is 00.0078125 per decimal increase but for display purposes
          we will round up or down to only display up to 100th degree accuracy
          */
          let celValRaw = (decVal*adcCelciusScalar);
          let celVal = celValRaw.toFixed(2);
          let farValRaw = (celVal*1.8000 +32);
          let farVal = farValRaw.toFixed(2);

          this.setState({ adcValue: rawData });
            this.setState({ celciusVal: celVal });
            this.setState({ farenheitVal: farVal });
            //take the base x64 string and convert into decimal then far/cel

          // eslint-disable-next-line no-catch-shadow
          } catch (error) {
            console.log(error.message);
          }
        }
      }
    );
  }`

meridPL, 2019-08-02:
Thank you for answer. Your advice very helpful for me. I didn't know manufacture data contains that infomation. I'm just wondering what function getDecFrom64 is.

dariuszseweryn, 2019-08-02:
https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-Beacons

Kosik-Chris, 2019-08-02:
getDecFrom64 is just a function I wrote to decode the x64 string to a decimal result. I updated it from the last code I put up in that it takes in the whole 64 bit string. Basic concept is to examine each x64 character, look at the ascii value, and get the equivalent x64 value (table lookup). I then just do some of the math to put it back together via the multiplications of 64^3, 64^2, 64 for each of the char digits. 
Example: DIA= is our x64 string, get the x64 value for D*64^3, get x64 value for I*64^2 etc.. Hope it helps!

closed 2019-10-08 by stale[bot]