dotintent/react-native-ble-plx #515: Scanning rate

Kosik-Chris, 2019-07-31:
Is there any way to get some numbers behind the "low latency" scan mode option? I am probably not passing on the options into 

`manager.startDeviceScan(null, {options: ScanMode.LowLatency}, (error, device) => {

if(error) {
console.log(error.message);
return
}`

My observation from logkitty is there is a device scan about once every second. 

Additionally, are there plans in the future to allow finer hardware level control of scanning through this library?

Kosik-Chris, 2019-08-01:
Update: Fixed some of my code.. Still notice about a 3 second window delay between scan results coming in. 8 values at a time coming in at a rate that corresponds well with mcu tx rate of 100ms 

Updated code:
`this.manager.startDeviceScan(
      [tiServiceID],
      ScanOptions,
      (error, device) => {
        if (error) {
          console.log(error.message);
          return;
        }`

Results I am seeing in logkitty:
8 tx packets read (softly) with Date.now() with an average interval of 100ms followed by a delay of 3316 ms (average 3 seconds) before continuing scan results are returned.

I still have a question regarding whether this library or rxAndroidBLE has any plans of allowing any hardware control beyond the defined modes Low_Latency etc.

dariuszseweryn, 2019-08-02:
> Is there any way to get some numbers behind the "low latency" scan mode option?

Yes, [here](https://stackoverflow.com/questions/47790910/rxandroidble-periodic-advertising-scan/47793092#47793092).

> Additionally, are there plans in the future to allow finer hardware level control of scanning through this library?

This is already about how much you can get from the system API

Kosik-Chris, 2019-08-02:
Thank you so much!

closed 2019-08-02 by Kosik-Chris