dotintent/react-native-ble-plx #516: Question: Beacon support

farzd, 2019-08-02:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x ] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Is there any reason this cant identify beacons ? i've tried scanning with the beacon UUID. Leaving the array as null identifies my Mac bluetooth address only. Testing it on iOS 12/iPhone 7

dariuszseweryn, 2019-08-02:
> Is there any reason this cant identify beacons ?

Yes, iOS uses a completely separate API for Beacons

I will try to add a wiki page for this topic in the future

closed 2019-08-02 by dariuszseweryn

dariuszseweryn, 2019-08-02:
https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-Beacons

farzd, 2019-08-03:
Thanks for the speedy reply