dotintent/react-native-ble-plx #518: Feature Request: Add support for new iOS 13 bluetooth permission/authorization feature

bitcrumb, 2019-08-09:
## Current Behavior

* The library should properly emit an "unauthorized" state when the user is not allowed to use bluetooth. However there is now way to determine through the library why this is. In iOS 13 the user can explicitly deny or approve bluetooth permission, since a dialog will be displayed by default for all new, updated *and* existing apps.

## New behavior

There should be a way to check the authorization status (via [CBManagerAuthorization](https://developer.apple.com/documentation/corebluetooth/cbmanagerauthorization?language=objc)) when bluetooth manager reports the "unauthorized" state to determine of a user denied the permission or not (and allow us to act accordingly in the UI).

ultimate5289, 2019-08-27:
I also experienced the same issue on iOS 13

ntrannghia, 2019-08-27:
I also experienced the same issue on iOS 13

ntrannghia, 2019-08-27:
hi @bitcrumb, I found out that BleManager.onStateChange can emit "Unauthorized" status

bitcrumb, 2019-08-28:
@ntrannghia Yes, that's what I explained in my issue. I rephrased a bit to make this more clear. What I *am* talking about however, is that there is no way to retrieve the reason for this "unauthorized" state. This is communicated through a a new API availablein iOS 13. The CBManagerAuthorization enum. This part has not been incorporated yet.

Also, the current version of ble-plx does not compile on the beta Xcode 11.

RWOverdijk, 2019-09-05:
@bitcrumb Yes, I notice the same.

> node_modules/react-native-ble-plx/ios/RxBluetoothKit/RxCBPeripheral.swift:364:121: 'canSendWriteWithoutResponse' is only available in iOS 11.0 or newer

bitcrumb, 2019-09-05:
There's a PR opened that presumably fixes this: https://github.com/Polidea/react-native-ble-plx/pull/527

bitcrumb, 2019-09-13:
Yes idd, I noticed that the PR has merged. Thanks for that :)
I think the only thing that remains now is indeed to have access to the enum containing the reason for being unauthorised to use bluetooth. This can come in handy to display certain specific UI/feedback to the user.

closed 2021-09-14 by bitcrumb