dotintent/react-native-ble-plx #519: Write hex command

sweebee, 2019-08-09:
I've received the BLE documentation of a adevice. I can setup wifi with it.

I can write the ssid to it and read it. But to join the network I need to send a command. But the device does nothing. I dont know how to send the values. These are the options:

command: ```0x02```
description: Initiate WLAN Join

But nothing happens. I also tried ```0x03``` which shout set it to WPS mode and the light on the device should start blinking very fast. But it doesn't.

How Should i send these command values to the device?

I've tried:

```
let val = base64.encode('0x03')
characteristic.writeWithoutResponse(val)
```
```
let val = base64.encode('3')
characteristic.writeWithoutResponse(val)
```
```
let val = base64.encode('000003')
characteristic.writeWithoutResponse(val)
```

dariuszseweryn, 2019-08-09:
You did not check [Wiki](https://github.com/Polidea/react-native-ble-plx/wiki/=--FAQ:-Passing-And-Retrieving-Of-Characteristic-Value), didn't you?

closed 2019-08-09 by dariuszseweryn

rhit-dingb, 2022-03-23:
Try to first check isWritable to see if the value is actually writable, then try to use the writeWithResponse, and use the returned characteristic in a promise and change the value there. I faced the same issue as you do while using without response but works with response. Good Luck!