dotintent/react-native-ble-plx #520: RN 60.4 Build Error

msalasSonatas, 2019-08-09:
Is this library compatible with latest RN version (60.4)?
I keep getting a react native native module cannot be null error when I try to setup this package in my project.

I was able to fix this by downgrading RN version to 59.3, but I just want to be sure If it's just not compatible with 60.4 yet, or I'm missing something.

izaxon, 2019-08-22:
I get the same errors with react-native 0.60.5. 
```
> Task :react-native-ble-plx:compileReleaseJavaWithJavac FAILED
/Users/vsts/agent/2.155.1/work/1/s/node_modules/react-native-ble-plx/android/src/main/java/com/polidea/reactnativeble/converter/RxBleScanResultConverter.java:3: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/vsts/agent/2.155.1/work/1/s/node_modules/react-native-ble-plx/android/src/main/java/com/polidea/reactnativeble/wrapper/Characteristic.java:5: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/vsts/agent/2.155.1/work/1/s/node_modules/react-native-ble-plx/android/src/main/java/com/polidea/reactnativeble/wrapper/Service.java:5: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
...
```
How can this be solved?

msalasSonatas, 2019-08-22:
> I get the same errors with react-native 0.60.5.
>
> ``` [...]

I wasn't able to resolve it on React Native versions above 0.59.3, I think thats the latest version that this library supports.

Cierpliwy, 2019-08-27:
It's mentioned in README file:

5. If you are using AndroidX, then for the time being you need to convert import statements in the Android library with [jetifier](https://www.npmjs.com/package/jetifier). These steps apply for all react-native packages, which are during the transition period:

- `npm install --save-dev jetifier`
- Run `jetify` script after `npm install`. You can do it by adding "postinstall" script to the `package.json` file:
  ```json
   ...
   "postinstall": "npx jetify",
   ...
  ```

Did you try that?

jeremywho, 2019-09-13:
If you're using yarn workspaces, you may need to add the library in the nohoist section of the projects package.json.:

```
"workspaces": {
        "nohoist": [
...
            "react-native-ble-plx",
            "react-native-ble-plx/**",
...
        ]
    }
```

closed 2019-11-19 by stale[bot]

rk85097, 2019-12-11:
Any solution for that? I have the same issue.

Reachip, 2019-12-16:
As same for me with this commands :

```
react-native link react-native-ble-plx 
npm run postinstall; npm run android
```

ToriToriToriTori, 2021-07-14:
Same with me. Has anyone fixed this?