dotintent/react-native-ble-plx #522: writeCharacteristicWithResponseForService the promise always returns "Operation was rejected"

Chourioz, 2019-08-10:
## Prerequisites
React Native 0.60.4
React Native Ble Plx 1.0.3
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question
Hello, I'm develop application to connect with smart bands, but when I write characteristic (00002a37-0000-1000-8000-00805f9b34fb) of heart rate service (0000180d-0000-1000-8000-00805f9b34fb), the promise always returns error and message is "Operation was rejected".

### My code is:

```javascript
const numberBuffer = Buffer.alloc(2);
numberBuffer.writeUInt16LE(1, 0);
console.warn(numberBuffer.toString('base64'))
const characteristic = await device.writeCharacteristicWithResponseForService(
   service, characteristicW, numberBuffer.toString('base64')
)
```

Cierpliwy, 2019-08-27:
Characteristic 0x2A37 in service 0x180D is notify only, [it can't be read.](https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Services/org.bluetooth.service.heart_rate.xml)

Marvedog, 2019-08-27:
### Addendum: 
How do you notify a characteristic that is notifiable only? If it isn't write-able, the promise always returns "Operation was rejected" (using the approach from the docs).

closed 2019-09-13 by Cierpliwy

akram-beleaf, 2024-05-16:
Hello Everyone 
I'm using an ESP32 and want to connect it with my app using bluetooth. 
I'm also facing similar issue. I'm trying writeCharacteristicWithResponseForService and getting the same error 

====>   **BleError:Operation was rejected**

Please help

here is my code for reference

`import React, { useEffect, useState } from "react";
import { Button, PermissionsAndroid, Platform, Text } from "react-native"
import { BleManager } from "react-native-ble-plx";
import { Buffer } from 'buffer'

const BLEServiceUUID = "6e400001-b5a3-f393-e0a9-e50e24dcca9e";
const BLEReadUUID = "6e400002-b5a3-f393-e0a9-e50e24dcca9e";
const BLEWriteUUID = "6e400003-b5a3-f393-e0a9-e50e24dcca9e";
const deviceID = "D8:BC:38:F9:0C:2A";

const bleManager = new BleManager();

const BleSerialComponent = () => {
    const [device, setDevice] = useState(null);

    const requestBluetoothPermission = async () => {
        if (Platform.OS === 'ios') {
            return true
        }
        if (Platform.OS === 'android' && PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION) {
            const apiLevel = parseInt(Platform.Version.toString(), 10)

            if (apiLevel < 31) {
                const result = await PermissionsAndroid.requestMultiple([
                    PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
                    PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
                ])
                return result
            }
            if (PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN && PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT) {
                const result = await PermissionsAndroid.requestMultiple([
                    PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
                    PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
                    PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION
                ])

                return (
                    result['android.permission.BLUETOOTH_CONNECT'] === PermissionsAndroid.RESULTS.GRANTED &&
                    result['android.permission.BLUETOOTH_SCAN'] === PermissionsAndroid.RESULTS.GRANTED &&
                    result['android.permission.ACCESS_FINE_LOCATION'] === PermissionsAndroid.RESULTS.GRANTED
                )
            }
        }
        return false
    }

    useEffect(() => {

        requestBluetoothPermission()
        return () => {
            device && device.cancelConnection();
        };
    }, []);

    const handleWrite = async () => {
        let data = Buffer.from('Hello', 'utf-8').toString('base64')
        console.log(data);
        const services = await device.discoverAllServicesAndCharacteristics();
        if (device) {
            try {
[... 110 lines cut ...]
        //   // Failure code
        //   console.log(error);
        // });

    }

    return (
        <>
            <Text>{device ? "Connected to:" + (device?.name || '') + "--"(device?.id || '') : 'No device connected'}</Text>
            {device ? (
                <Button title="Disconnect" onPress={disconnectDevice} />
            ) : (
                <Button title="Scan and Connect" onPress={scanAndConnect} />
            )}
            <Button title={'write'} onPress={handleWrite} />
            <Button title={'read'} onPress={handleRead} />
            <Button title={'checkConnection'} onPress={checkConnection} />
            <Button title={'connectDevice'} onPress={connectDevice} />
        </>
    )
};

export default BleSerialComponent;`

after connecting to the esp32 when I press handleWrite the I get this error.