dotintent/react-native-ble-plx #523: Tutorial failing to build with 'modulemap'

yslee9311, 2019-08-12:
## Expected

Xcode should build the project clearly.

## Current

Get Error :

```
<unknown>:0: error: module map file '/Users/me/Library/Developer/Xcode/DerivedData/bleplx-gevvblkqzrxbjediwbytdxcmpufo/Build/Products/Debug-iphonesimulator/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found
<unknown>:0: error: module map file '/Users/me/Library/Developer/Xcode/DerivedData/bleplx-gevvblkqzrxbjediwbytdxcmpufo/Build/Products/Debug-iphonesimulator/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found
2 errors generated.
<unknown>:0: error: failed to emit precompiled header '/Users/me/Library/Developer/Xcode/DerivedData/bleplx-gevvblkqzrxbjediwbytdxcmpufo/Build/Intermediates.noindex/PrecompiledHeaders/bleplx-Bridging-Header-swift_YXQHX0LCXQ6A-clang_1Y5VOCRBO1JAC.pch' for bridging header '/Users/me/Desktop/bleplx/ios/bleplx-Bridging-Header.h'
```

I installed it according to the manual and a problem occurred.
I don't know how to fix it.
How can i solve this problem?

chazguerrero, 2019-08-20:
i just spent a couple nights on trying to figure this out. after you follow the instructions open project.xcworkspace and build from that not the xcodeproj. worked for me after that 
@yslee9311

Cierpliwy, 2019-08-27:
@chazguerrero Did you see any changes in your project's files after doing your last step? I would like to know if something changed.

gregogalante, 2019-08-30:
Getting the same problem with these dependencies:

```json
{
    "react": "16.8.6",
    "react-native": "0.60.4",
    "react-native-ble-plx": "^1.0.3",
    "react-native-gesture-handler": "^1.4.1",
    "react-native-json-tree": "^1.2.0",
    "react-navigation": "^3.12.1"
}
```

Cierpliwy, 2019-09-13:
Make sure to go to the `ios` folder and run `pod update` inside it. Remember to open the workspace file in the XCode (not a project). Do full clean/build cycle.

Vergil0327, 2019-11-04:
I follow the installation step and open xcode with .xcodeproj.
After clean & build, I got the same error, too.

`Swift Compiler Error`

`<unknown>:0: error: module map file '/Users/user/Library/Developer/Xcode/DerivedData/rnProject-ewpnsocexbnhsyfuyaequipzsyfp/Build/Products/Debug-iphoneos/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found`

`<unknown>:0: error: module map file '/Users/user/Library/Developer/Xcode/DerivedData/rnProject-ewpnsocexbnhsyfuyaequipzsyfp/Build/Products/Debug-iphoneos/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found
2 errors generated.`

`<unknown>:0: error: failed to emit precompiled header '/Users/user/Library/Developer/Xcode/DerivedData/rnProject-ewpnsocexbnhsyfuyaequipzsyfp/Build/Intermediates.noindex/PrecompiledHeaders/rnProject-Bridging-Header-swift_29FFIXLGOXG6T-clang_39HZT4NQHDNJ9.pch' for bridging header '/Users/user/Documents/rnProject/ios/rnProject-Bridging-Header.h'`

closed 2020-01-11 by stale[bot]

gegana, 2020-01-31:
> Make sure to go to the `ios` folder and run `pod update` inside it. Remember to open the workspace file in the XCode (not a project). Do full clean/build cycle.

I'm just going to add on to this point here that if you are running using xcodebuild, you also need to target the workspace `-workspace` instead of `-project`.

sureshmss2007, 2020-05-14:
For me also it worked by pointing out workspace instead of project. 
http://helpinterview.com/?q=content/fatal-error-module-map-file-not-found-xcode

Jerry-Lau, 2020-11-22:
why should u open the .xcworkspace instead of the .xcproj

dariuszseweryn, 2020-11-22:
> why should u open the .xcworkspace instead of the .xcproj

Cocoapods

MonchiLin, 2022-04-28:
> > 确保转到该`ios`文件夹​​并`pod update`在其中运行。请记住在 XCode（不是项目）中打开工作区文件。进行完整的清洁/构建周期。
> 
> 我只是在这里补充一点，如果你使用 xcodebuild 运行，你还需要定位工作区`-workspace`而不是`-project`.

Save my life.