dotintent/react-native-ble-plx #526: Handling case where BLE is turned on before opening app on iOS

henry132109, 2019-08-27:
I see that the documentation mentions that:
> When iOS application launches BLE stack is not immediately available and we need to check its status. To detect current state and following state changes we can use onStateChange() function

I'm using this snippet in **componentDidMount**:
```
if (Platform.OS === 'ios') 
  const subscription = this.manager.onStateChange((state) => {
    if (state === 'PoweredOn'){
      //do my stuff
      subscription.remove();
    }
  })
} else {
  this.manager.state().then((state)=>{
    if(state === 'PoweredOn'){
      //do my stuff
    }else{
      this.manager.enable().then(()=>{
        //do my stuff
      });
    }
  })
}
```

But my stuff doesn't fire when BLE was turned on before opening the app.
How do I ensure my event can be fired only once?

Cierpliwy, 2019-08-27:
If you really want to check it once you can use [`state()`](https://polidea.github.io/react-native-ble-plx/#blemanagerstate) function directly. However, I would recommend to use [`onStateChange(listener, emitCurrentState`)](https://polidea.github.io/react-native-ble-plx/#blemanageronstatechange) with last parameter set to true.

henry132109, 2019-08-27:
Thank you for your quick reply!
I'm referring to my colleague and see if it works out :)

henry132109, 2019-08-28:
It worked like charm, dziękuję!

closed 2019-08-28 by henry132109