dotintent/react-native-ble-plx #528: startDeviceScan not returning any values in Android ...

nachten, 2019-09-03:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ X] I am running the latest version
- [ X] I checked the documentation and found no answer
- [ X] I checked to make sure that this issue has not already been filed

## Expected Behavior

Getting results from "startDeviceScan" method in android

## Current Behavior

Previously it worked on an android device but after an upgrade of react-native from 0.60.4 to 0.60.5 it does not work anymore ... Downgrading back to 0.60.4 doesn't solve my problem ... 
Everything works fine in IOS.

And yes permissions are set in the APP ... 

* Library version: 1.0.3
* Platform: Android.
* Contents of the `package.json` file:

```json
{
  "name": "some_app",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint .",
  },
  "rnpm": {
    "assets": [
      "src/assets/fonts"
    ]
  },
  "dependencies": {
    "buffer": "^5.2.1",
    "jwt-decode": "^2.2.0",
    "lodash": "^4.17.15",
    "moment": "^2.24.0",
    "react": "16.8.6",
    "react-native": "0.60.4",
    "react-native-ble-plx": "^1.0.3",
    "react-native-device-info": "^2.2.2",
    "react-native-elements": "^1.1.0",
    "react-native-gesture-handler": "^1.3.0",
    "react-native-splash-screen": "^3.2.0",
    "react-native-vector-icons": "^6.6.0",
    "react-navigation": "^3.11.1",
    "react-redux": "^7.1.0",
    "reconnecting-websocket": "^4.1.10",
    "redux": "^4.0.4",
    "redux-saga": "^1.0.5",
    "socket.io-client": "^2.2.0"
  },
  "devDependencies": {
    "@babel/core": "7.5.5",
    "@babel/runtime": "7.5.5",
    "@react-native-community/eslint-config": "0.0.3",
    "babel-jest": "24.8.0",
    "eslint": "6.0.1",
    "jest": "24.8.0",
    "jetifier": "^1.6.4",
    "metro-react-native-babel-preset": "0.54.1",
    "react-native-dotenv": "^0.2.0",
    "react-test-renderer": "16.8.6"
  },
  "jest": {
    "preset": "react-native"
  }
}

```

* Formatted code sample or link to a repository:

```js
 manager.startDeviceScan(null, null, (error, scannedDevice) => {

            console.info('swing');
            if (error) {
                console.info(error);
                emit([error, scannedDevice]);
                return;
            }

            //console.info(scannedDevice);

            // this option works both on android & ios
            if (scannedDevice.serviceUUIDs && scannedDevice.serviceUUIDs.indexOf('ef680100-9b35-4933-9b10-52ffa9740042') > -1)
                emit([error, scannedDevice]);

        });
```

sabinbogati1, 2019-09-08:
I am facing same problem.. How to I solve this problem?

PhamMinhKha, 2019-09-10:
+ 1

tedcurrent, 2019-09-11:
I had a similar issue, but I did a battery reset for the BLE devices, which caused them to advertise themselves for scanning.

Cierpliwy, 2019-09-13:
Did you do following things:
- Enable coarse/fine location permission.
- Enable location services (GPS icon).
?

nachten, 2019-09-13:
Yes permissions are all set correctly

rk85097, 2019-11-06:
Same issue here - any solution?

ohmybrew, 2019-11-27:
Same issue here. Android 10. Was working previous on 9 :/

dariuszseweryn, 2019-11-27:
Android 10 requires `ACCESS_FINE_LOCATION` permission to return scan results.

closed 2019-11-27 by dariuszseweryn

ScreamZ, 2019-12-19:
@dariuszseweryn Can this be added in the doc with a sample. So for android < 6 you don't need any permission for > 6 can we just set FINE LOCATION or we also need COARSE ?

dariuszseweryn, 2020-07-01:
[This info is available on the related wiki page](https://github.com/Polidea/react-native-ble-plx/wiki/Bluetooth-Scanning)