dotintent/react-native-ble-plx #541: Unknown error occurred. This is probably a bug! Check reason property.

wenchao1020, 2019-09-17:
Hello.
  I used `monitorCharacteristicForDevice` to get data from device, but it always show an exception like that:
 ```javascript
{
    "message":"Unknown error occurred. This is probably a bug! Check reason property.",
    "errorCode":0,
    "attErrorCode":null,
    "iosErrorCode":null,
    "androidErrorCode":null,
    "reason":"rx.exceptions.MissingBackpressureException",
    "name":"BleError"
}
```
and the connect is lost, here is my code
```javascript
monitor = (index) => {
        let transactionId = 'monitor';
        this.blueDataBuffer = ''
        this.monitorListener = BluetoothManager.manager.monitorCharacteristicForDevice(BluetoothManager.peripheralId,
            BluetoothManager.nofityServiceUUID[index],BluetoothManager.nofityCharacteristicUUID[index],
            (error, characteristic) => {
                if (error) {
                    console.log('monitor fail:',JSON.stringify(error));  
                    this.addMsg(JSON.stringify(error))  
                    this.alert('monitor fail: ' + error.reason);      
                }else{
                   
                    let newValue = Buffer.from(characteristic.value, 'base64').toString('hex')
                    console.log(newValue)
                    this.blueDataBuffer = `${this.blueDataBuffer}${newValue}`
                    
                    // if(this.blueDataBuffer.length > 800){
                    //     if(!this.hasDeviceInfo) {
                    //         this.getDeviceInfo()
                    //     } else {
                    //         this.dealData(newValue)
                    //     }
                    // }
                }
            }, transactionId)
    } 
```

closed 2019-09-25 by wenchao1020

SaltedBlowfish, 2020-03-23:
@wenchao1020 Why did you close this? Did you find a solution? I'm running into the same issue.

Symous, 2023-09-12:
@wenchao1020  @SaltedBlowfish Hi, friend, how to solve this?

LLMoreira, 2024-02-04:
Eu resolvi com isso:

import React, { useState, useEffect } from 'react';
import { View, Text, FlatList, Platform } from 'react-native';
import { BleManager, Device } from 'react-native-ble-plx';
import { check, request, PERMISSIONS, RESULTS } from 'react-native-permissions';

const BluetoothScanner: React.FC = () => {
  const [devices, setDevices] = useState<Device[]>([]);
  const manager = new BleManager();

  useEffect(() => {
    const requestPermissions = async () => {
      if (Platform.OS === 'android') {
        const locationPermissionStatus = await check(PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION);
        if (locationPermissionStatus !== RESULTS.GRANTED) {
          const Location = await request(PERMISSIONS.ANDROID.ACCESS_FINE_LOCATION);
          const SCAN = await request(PERMISSIONS.ANDROID.BLUETOOTH_SCAN);
          const CONNECT = await request(PERMISSIONS.ANDROID.BLUETOOTH_CONNECT);
          if (Location !== RESULTS.GRANTED && SCAN !== RESULTS.GRANTED && CONNECT !== RESULTS.GRANTED) {
            console.warn('Permission denieds');
            return;
          }
        }
      }

      const subscription = manager.onStateChange((state) => {
        if (state === 'PoweredOn') {
          startScan();
        }
      }, true);

      return () => {
        subscription.remove();
      };
    };

    requestPermissions();
  }, []);

  const startScan = () => {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.error(error);
        return;
      }

      if (device && !devices.some((d) => d.id === device.id)) {
        setDevices((prevDevices) => [...prevDevices, device]);
      }
    });
  };

  return (
    <View>
      <Text>Bluetooth Devices:</Text>
      <FlatList
        data={devices}
        keyExtractor={(item) => item.id}
        renderItem={({ item }) => <Text>{item.name || 'Unknown Device'}</Text>}
      />
    </View>
  );
};

export default BluetoothScanner;

newlomar, 2024-10-15:
Hey everyone!

@wenchao1020 how did you fix this?