dotintent/react-native-ble-plx #543: Ble stops sending notifications in background

mitaxe, 2019-09-19:
I stacked with the issue in android when my device is in the background mode or locked it stops getting notification about statuses through the Bluetooth in some period of time. I have to keep alive connection all the time. Haven't found this issue in iOS. Is this a known issue?

dariuszseweryn, 2019-09-19:
Android stops backgrounded applications unless you keep them foreground via a `foreground service` for instance (you can google it).

mitaxe, 2019-09-22:
Would you be so kind to suggest any library that could help to resolve the issue?
What do you think about [React Native Background Geolocation](https://github.com/mauron85/react-native-background-geolocation)?

Or there is some other smart solution?

closed 2019-11-28 by stale[bot]