dotintent/react-native-ble-plx #548: Must call 'device.discoverAllServicesAndCharacteristics()' multiple times

tomdaniel0x01, 2019-09-28:
## Prerequisites

- [ YES ] I am running the latest version
- [ YES ] I checked the documentation and found no answer
- [ YES ] I checked to make sure that this issue has not already been filed

## Expected Behavior

I expect to only have to call 'device.discoverAllServicesAndCharacteristics()' once per device as documentation states, "Discovery of services and characteristics is required to be executed only once."

## Current Behavior

If I disconnect from a device and then reconnect I get the following error, "BleError: Service {actual service uuid here} for device ? not found". However, I do not get this error if I call 'device.discoverAllServicesAndCharacteristics()' again after connecting.

I also tried saving the device object and using from that cache instead of grabbing it from manager again. This did not resolve the problem.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. manager.startDeviceScan()
2. device.connect()
3. device.discoverAllServicesAndCharacteristics()
4. device.cancelConnection()
5. device.connect()
6. device.writeCharacteristicWithResponseForService()

### Context

* Library version: 1.1.0
* Platform: Android

dariuszseweryn, 2019-12-02:
Thank you for the report. It is a bug in documentation. I have created a PR that should fix it.
The discovery must be executed at least (usually exactly) once per connection to interact with characteristics.

tomdaniel0x01, 2019-12-02:
@dariuszseweryn Thanks for the heads up!

PR #583 (2020-01-09): Fixed intro information about service discovery
Closes #548
closed 2019-12-02 by tomdaniel0x01