dotintent/react-native-ble-plx #551: Create BleManager() without a native component

dschinkel, 2019-10-04:
I'm abstracting ble logic out of my native components and pushing that down to be functional behavior that lives out side my component.  

I created a function in a JS Custom Component of mine that tries to use the manager:
```
	const manager = new BleManager();

	function scanForDevice(){
		manager.startDeviceScan(null, null, (error, device) => {
			return device;
		});
	}
```

When I run my unit test, it calls this function eventually and tries to use that manager instance, but I get the famous `Cannot read property 'createClient' of undefined` because the manager is undefined.

I went through the docs, step by step and still I have this issue.  Is it just not possible to do that without a React Native component?

mlisik, 2019-10-30:
Can you post the contents of the failing unit test (imports, mocks, test itself)? Perhaps it would be fixed by simply mocking out the dependency like so: `jest.mock('react-native-ble-plx');`?

closed 2020-01-05 by stale[bot]

riegaz, 2020-02-15:
I do have the same issue. Can anyone help me? I'm completely new.

[image]

Cierpliwy, 2020-02-18:
It means that BLE native module is not properly loaded. Please recheck instruction steps and make sure you link your native dependencies from the root of the project.

ouya99, 2021-06-01:
Any update on this issue? cannot get past the error on Expo

aarthigovardhanan88, 2021-06-17:
Any updates on this? I am also facing the same issue

kriit24, 2022-02-20:
same issue

KyrolosMagdy, 2023-03-01:
Same issue

Moyando, 2023-03-28:
same issue

lokacoding, 2023-04-19:
same issue

gokhantiknaz, 2023-05-13:
same issue.

DevKalebe, 2023-05-19:
mesmo problema.

mathiasnohall, 2023-07-11:
same issue

Mottion, 2023-07-19:
mesmo problema.

Naveen-g09, 2024-04-30:
have anyone figured it out?

Naveen-g09, 2024-04-30:
hey @everyone, tis fixed my issue:
```{
  "expo": {
    "plugins": ["react-native-ble-plx"]
  }
}
```

MaghsoodTT, 2024-06-26:
I did these:
1- useing `npx expo run:android` instead of `npx expo run:android`
2- it only works with USB debugging, before run command connect device to the usb and check connection by `adb devices` 
3- `npx expo run:android` might help