dotintent/react-native-ble-plx #552: How to see if a device lost connection

keyz23, 2019-10-08:
Hi I'm using this library to connect to two external devices. I would know if there is a method to know, after that connection has been enstablished, if a device lost the BLE connection because for example is to far from the smartphone. 

It is possibile?

Thank you

diamanthaxhimusa, 2019-10-17:
@keyz23
`
  import { BleManager } from 'react-native-ble-plx';

  let manager = new BleManager();
  manager.onDeviceDisconnected(<device_id>, (error, device) => {
    if(error) {
      console.log(error);
    }
    console.log("Device is disconnected");
  });
`

keyz23, 2019-10-17:
@diamanthaxhimusa Thank you for your reply. I would explain better my problem to understand how to use in a better way your code. 

I have a page in which I'm doing the scan and Connection(1). Then I pass in another page where the user can choose some activities to do(2) and in the third page I do the discoverAllServicesAndCharacteristics and other interaction with ble (3).

Now sometimes in the page (2) I receive this Alert: BleError device (id number) was disconnected.

And I can't recover the ServicesAndCharacteristics. 

But other times it works perfectly (the devices are one near the other). 

Can I use your function to constrast this lost of connection? Thank you

closed 2019-12-23 by stale[bot]