dotintent/react-native-ble-plx #553: Issue with paring after forget device

shamprasadrh, 2019-10-09:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

When we connect to a device and write to encrypted characteristics, we get the IOS pairing dialog and write will be successful. if the the paired device is removed using `Forget Device` option in IOS  setting and try to write to characteristic next time  it should again prompt for pairing and write should be successful. 

## Current Behavior

If device is removed using `Forget Device` option using the IOS settings, write operation always fails with promise rejection thereafter. 

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Do a scan, connect and discover the services and characteristics 
2. Write to the encrypted characteristic with pairing 
3. Forget device from the IOS bluetooth settings
4. Repeat steps 1 & 2.

* Library version: 1.1.0
* Platform: iOS.
* Platform logs (logcat/XCode): 
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

Error

```js
BleError: Characteristic xxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxx write failed for device xxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxx  and service xxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxx 
```

kevinresol, 2019-12-12:
This is also an issue on android

Cierpliwy, 2020-01-31:
I don't see any reason for it to not work. Can you send me more details? 
* Native logs from XCode / Android Studio
* Information about characteristic's permissions

closed 2020-04-07 by stale[bot]