dotintent/react-native-ble-plx #554: Bug: Unable to read packages larger than 185 bytes in iOS

josectobar, 2019-10-09:
## Prerequisites

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed

## Expected Behavior

I'm trying to read a characteristic that sends a JSON string as the value, this string is larger than 185 bytes.

## Current Behavior

I'm trying to read a characteristic that sends a JSON string as the value, this string is larger than 185 bytes which is not a problem with Android but it's being cut off when I read it in iOS.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Scan for devices. 
2. connect to device
3. Discover all services and characteristics for device.
4. send write characteristic with response for device logging into the device with a `username` and `password`.
5. send a read characteristic for device (both read and write use the same service and characteristic).
6. when I read the value, it's cut off and repeated 3 times. 

**the correct response that I get with Android:** 

**Base 64:**

``
eyJqc29ucnBjIjoiMi4wIiwicmVzdWx0IjpbMCx7InVidXNfcnBjX3Nlc3Npb24iOiJkMmYyZDVlYTIxOWIxMDhhZGEzM2Q4NGFiOGE0ZjRjYyIsInRpbWVvdXQiOjMwMCwiZXhwaXJlcyI6Mjk5LCJhY2xzIjp7ImFjY2Vzcy1ncm91cCI6eyJ4eHh4eC11c2VyIjpbInJlYWQiLCJ3cml0ZSJdfSwidWJ1cyI6eyJmd21hbiI6WyIqIl0sInh4eHh4LmFjcyI6WyJyZWJ1aWxkX3h4eHhfcHJvZmlsZXMiXSwieHh4eHguYXBwIjpbIioiXSwibmV0d29yayI6WyJyZWxvYWQiXSwibmV0d29yay5pbnRlcmZhY2UueHh4IjpbInN0YXR1cyJdLCJuZXR3b3JrLmludGVyZmFjZS54eHgiOlsic3RhdHVzIl0sInVjaSI6WyIqIl19LCJ1Y2kiOnsiKiI6WyJyZWFkIiwid3JpdGUiXX19LCJkYXRhIjp7InVzZXJuYW1lIjoiYWRtaW4ifX1dLCJpZCI6MX0=
``

**JSON string:**

``
{"jsonrpc":"2.0","result":[0,{"ubus_rpc_session":"d2f2d5ea219b108ada33d84ab8a4f4cc","timeout":300,"expires":299,"acls":{"access-group":{"xxxxx-user":["read","write"]},"ubus":{"fwman":["*"],"xxxxx.acs":["rebuild_xxxx_profiles"],"xxxxx.app":["*"],"network":["reload"],"network.interface.xxx":["status"],"network.interface.xxx":["status"],"uci":["*"]},"uci":{"*":["read","write"]}},"data":{"username":"admin"}}],"id":1}
``

**the response that I get with iOS:** 

**Base 64:**

``
eyJqc29ucnBjIjoiMi4wIiwicmVzdWx0IjpbMCx7InVidXNfcnBjX3Nlc3Npb24iOiJiOTBmYTAyZDk4NjI3MjdjMWY2ZWQyMThmYzQ1ODEyYyIsInRpbWVvdXQiOjMwMCwiZXhwaXJlcyI6Mjk5LCJhY2xzIjp7ImFjY2Vzcy1ncm91cCI6eyJ4eHh4eC11c2VyIjpbInJlYWQiLCJ3cml0ZSJdfSwidWJ1cyI6eyJmd21hbiI6W3sianNvbnJwYyI6IjIuMCIsInJlc3VsdCI6WzAseyJ1YnVzX3JwY19zZXNzaW9uIjoiYjkwZmEwMmQ5ODYyNzI3YzFmNmVkMjE4ZmM0NTgxMmMiLCJ0aW1lb3V0IjozMDAsImV4cGlyZXMiOjI5OSwiYWNscyI6eyJhY2Nlc3MtZ3JvdXAiOnsieHh4eHgtdXNlciI6WyJyZWFkIiwid3JpdGUiXX0sInVidXMiOnsiZndtYW4iOlt7Impzb25ycGMiOiIyLjAiLCJyZXN1bHQiOlswLHsidWJ1c19ycGNfc2Vzc2lvbiI6ImI5MGZhMDJkOTg2MjcyN2MxZjZlZDIxOGZjNDU4MTJjIiwidGltZW91dCI6MzAwLCJleHBpcmVzIjoyOTksImFjbHMiOnsiYWNjZXNzLWdyb3VwIjp7Inh4eHh4LXVzZXIiOlsicmVhZCIsIndyaXRlIl19LCJ1YnVzIjp7ImZ3bWFuIjpb
``

**JSON string:**

``
{"jsonrpc":"2.0","result":[0,{"ubus_rpc_session":"b90fa02d9862727c1f6ed218fc45812c","timeout":300,"expires":299,"acls":{"access-group":{"xxxxx-user":["read","write"]},"ubus":{"fwman":[{"jsonrpc":"2.0","result":[0,{"ubus_rpc_session":"b90fa02d9862727c1f6ed218fc45812c","timeout":300,"expires":299,"acls":{"access-group":{"xxxxx-user":["read","write"]},"ubus":{"fwman":[{"jsonrpc":"2.0","result":[0,{"ubus_rpc_session":"b90fa02d9862727c1f6ed218fc45812c","timeout":300,"expires":299,"acls":{"access-group":{"xxxxx-user":["read","write"]},"ubus":{"fwman":[
``

* Library version:1.1.0
* Platform: iOS.
* Platform logs XCode:
```
[RxBLEKit|VERB|15:55:51.878]: CentralManager(7522903616) didDiscover(peripheral: Peripheral(uuid: B2FF0335-498F-54CE-8360-FE992BD24D83, name: Optional("BlueZ 5.49")),
rssi: -53)
[RxBLEKit|DEBG|15:55:52.126]: CentralManager(7522903616) stopScan()
[RxBLEKit|DEBG|15:55:52.128]: CentralManager(7522903616) retrievePeripherals(
withIdentifiers: [B2FF0335-498F-54CE-8360-FE992BD24D83])
[RxBLEKit|DEBG|15:55:52.137]: CentralManager(7522903616) connect(
peripheral: Peripheral(uuid: B2FF0335-498F-54CE-8360-FE992BD24D83, name: Optional("BlueZ 5.49")),
options: nil)
[RxBLEKit|DEBG|15:55:53.288]: Peripheral(uuid: B2FF0335-498F-54CE-8360-FE992BD24D83, name: Optional("BlueZ 5.49")) discoverServices(
serviceUUIDs: nil)
[RxBLEKit|DEBG|15:55:53.925]: Peripheral(uuid: B2FF0335-498F-54CE-8360-FE992BD24D83, name: Optional("BlueZ 5.49")) didDiscoverServices(services
: Optional("[Service(uuid: B5DF559F-F530-4661-84C6-F3B9D5F76985, id: 7520873408)]"),
error: nil)
[... 69 lines cut ...]
                  },
                ],
              }),
            ).toString('base64'),
          )
        } catch (err) {
          console.error(err)
        }
        console.log({ write })

        const read = await manager.readCharacteristicForDevice(
          deviceUUID,
          jsonDataServiceUUID,
          jsonFormattedCharacteristicUUID,
        )
        if (read.value) {
          console.log(read)
          const responseJson = Buffer.from(read.value, 'base64').toString()
          console.log(responseJson)
        }
      }
    })
  }
  bluetooth()
```

tasqyn, 2019-11-06:
Hi @josectobar did able to solve this issue? I am also having these problem.

josectobar, 2019-11-06:
I did, the problem was related to the client. The Client was trying to send reads in one singe package, this was a problem for both android and iOS since it would exceed the MTU. We are now testing a different version that will send the reads in chunks, and we hope that this will solve this problem. My advice, check the client and make sure that it's sending packages within the MTU threshold.

josectobar, 2019-11-06:
@tasqyn ☝️

closed 2020-01-12 by stale[bot]

Gargamil, 2020-05-29:
[iOS has a fixed MTU of 185](https://github.com/chrisc11/ble-guides/blob/master/ios-mtu.md
)