dotintent/react-native-ble-plx #555: react_native_ble_plx_swift.modulemap not found

dschinkel, 2019-10-11:
Can't seem to get past this error when I build my ios project.  I've tried running `react-native link` but no luck.  Doesn't React Native auto link now anyway?  Out of ideas of what to try to get rid of this compile issue

Cierpliwy, 2019-10-11:
@dschinkel Make sure to update to the latest Cocoapods as there was a bug introduced: https://github.com/airbnb/lottie-ios/issues/859.
Additionally make sure that in Build Settings -> Swift Compiler - Language -> Swift Language Version is set to Swift 4.x.
Let me know if anything helps.

dschinkel, 2019-10-14:
thanks will try that out.

Vergil0327, 2019-11-04:
I follow the installation step and open xcode with .xcodeproj.
After clean & build, I got the same error, too.
My Swift Language Version is set to Swift 4.2

`Swift Compiler Error`

`<unknown>:0: error: module map file '/Users/user/Library/Developer/Xcode/DerivedData/rnProject-ewpnsocexbnhsyfuyaequipzsyfp/Build/Products/Debug-iphoneos/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found`

`<unknown>:0: error: module map file '/Users/user/Library/Developer/Xcode/DerivedData/rnProject-ewpnsocexbnhsyfuyaequipzsyfp/Build/Products/Debug-iphoneos/react-native-ble-plx-swift/react_native_ble_plx_swift.modulemap' not found
2 errors generated.`

`<unknown>:0: error: failed to emit precompiled header '/Users/user/Library/Developer/Xcode/DerivedData/rnProject-ewpnsocexbnhsyfuyaequipzsyfp/Build/Intermediates.noindex/PrecompiledHeaders/rnProject-Bridging-Header-swift_29FFIXLGOXG6T-clang_39HZT4NQHDNJ9.pch' for bridging header '/Users/user/Documents/rnProject/ios/rnProject-Bridging-Header.h'`

And I have a question about installation.
Do I need `react-native link react-native-ble-plx` when react native cli (0.60+) uses autolinking for dependencies?

closed 2020-01-11 by stale[bot]