dotintent/react-native-ble-plx #557: Data formatting for write

heatherSnepenger, 2019-10-15:
I'm trying to write, and getting this error:
`BleError: Characteristic * write failed for device * and service *`

I believe it's an issue with the formatting of the write data. The bluetooth device is expecting a data array, specifically something like this:  
[ 0, 0, 173, 0, 129, 164, 116, 121, 112, 101, 170,  100,  105, 97, 103,  110, 111, 115, 116, 105 ]

It seems the lib wants the above to converted to a string. When I convert it to a base64 string, the bluetooth device fails to write. Anyone know how to write a data array?

wfischer42, 2019-10-20:
I'm having the same issue. Specifically, I need to send UINT16 or raw binary, so I'm not able to transmit the data the device expects.

mazenchami, 2019-11-06:
@heatherSnepenger my hardware is also initially expecting a byte array, and I haven't had any issues with my example below.
Here is how i'm doing mine (keep in mind this could be specific to my device):
```
import { Buffer } from 'buffer';
...
const readSingleWifiShortHex = [parseInt('0xAB', 16), parseInt('0x01', 16), parseInt('0x01', 16)];
const readSingleWifiShortBase64 = new Buffer(readSingleWifiShortHex).toString('base64');
return await device.writeCharacteristicWithResponseForService(serviceUUID, characteristicUUID, readSingleWifiShortBase64, singleWifiShortTransactionId)
            .then(async characteristic => await console.log('characteristic', characteristic))
...
```
for `readSingleWifiShortHex`, index 0 = the command, 1 = length, 2 = payload.

does that help?

closed 2020-01-12 by stale[bot]

ayberkgerey, 2021-02-07:
@mazenchami Are you creating your own transactionId value or fetching from somewhere ?

mazenchami, 2021-02-08:
@ayberkgerey I am creating my own `transactionId` that way I can use it to read later on in my code/file.