dotintent/react-native-ble-plx #561: I want to know if this package is compatible with IOS13 and Android 2.2

Rebaiahmed, 2019-10-29:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.

harveyconnor, 2019-11-07:
ios13 yes.
2.2 no

mazenchami, 2019-11-07:
@harveyconnor @Rebaiahmed please reference this issue logged as it seems to be breaking for iOS13 (only when compiled with Xcode 11.2) - https://github.com/Polidea/react-native-ble-plx/issues/565

harveyconnor, 2019-11-07:
@mazenchami update to xcode 11.2.1 GM seed

closed 2020-01-14 by stale[bot]