dotintent/react-native-ble-plx #565: [iOS]: Crashed: com.facebook.react.BleClientManagerQueue EXC_BAD_ACCESS

mazenchami, 2019-11-04:
## Expected Behavior

When I initialize `new BleManager()`, the app shouldn't crash.

## Current Behavior

On opening of page that initializes BleManager, the app crashes

### Context

* Library version: `1.1.0`
* Platform: `iOS`
* React-Native Version: `0.59.4`
* Xcode: `11.2`

The error in the title is what I get from Crashlytics for either AdHoc or Release builds. Works as expected in debug.

I have put the code below in `constructor`, `componentDidMount`, & `componentWillMount` and still have the same error
```
let bleManager = new BleManager();
bleManager.onStateChange(state => this.setState({ bleState: state, }), true);
```

###react-native info
```React Native Environment Info:
    System:
      OS: macOS 10.15.1
      CPU: (8) x64 Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
      Memory: 295.31 MB / 16.00 GB
      Shell: 5.7.1 - /bin/zsh
    Binaries:
      Node: 11.0.0 - /usr/local/bin/node
      Yarn: 1.7.0 - /usr/local/bin/yarn
      npm: 6.4.1 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 13.2, DriverKit 19.0, macOS 10.15, tvOS 13.2, watchOS 6.1
      Android SDK:
        API Levels: 23, 24, 25, 26, 27, 28
        Build Tools: 23.0.1, 25.0.0, 25.0.2, 25.0.3, 26.0.2, 27.0.3, 28.0.3
        System Images: android-23 | Google APIs Intel x86 Atom_64, android-25 | Google APIs Intel x86 Atom, android-28 | Google Play Intel x86 Atom
    IDEs:
      Android Studio: 3.4 AI-183.6156.11.34.5522156
      Xcode: 11.2/11B52 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.8.3 => 16.8.3 
      react-native: 0.59.4 => 0.59.4 
    npmGlobalPackages:
      react-native-cli: 2.0.1```

Balake, 2019-11-05:
Getting something similar as well. Haven't found a fix yet.

Balake, 2019-11-06:
Here's my crash log if it helps any:

```
Crashed: com.facebook.react.BleClientManagerQueue
0  libswiftCore.dylib             0x1b88e2044 _swift_initClassMetadataImpl(swift::TargetClassMetadata<swift::InProcess>*, swift::ClassLayoutFlags, unsigned long, swift::TypeLayout const* const*, unsigned long*, bool) + 1688
1  libswiftCore.dylib             0x1b88e1f28 _swift_initClassMetadataImpl(swift::TargetClassMetadata<swift::InProcess>*, swift::ClassLayoutFlags, unsigned long, swift::TypeLayout const* const*, unsigned long*, bool) + 1404
2  myApp                          0x102c83174 type metadata completion function for ReplaySubject + 1976112
3  libswiftCore.dylib             0x1b88e8734 swift::MetadataCacheEntryBase<(anonymous namespace)::GenericCacheEntry, void const*>::doInitialization(swift::ConcurrencyControl&, swift::MetadataCompletionQueueEntry*, swift::MetadataRequest) + 268
4  libswiftCore.dylib             0x1b88dfc64 swift_getGenericMetadata + 1160
5  myApp                          0x102c831bc type metadata accessor for ReplaySubject + 1976184
6  libswiftCore.dylib             0x1b88f5adc (anonymous namespace)::DecodedMetadataBuilder::createBoundGenericType(swift::TargetContextDescriptor<swift::InProcess> const*, llvm::ArrayRef<swift::TargetMetadata<swift::InProcess> const*>, swift::TargetMetadata<swift::InProcess> const*) const + 248
7  libswiftCore.dylib             0x1b88f4de0 swift::Demangle::TypeDecoder<(anonymous namespace)::DecodedMetadataBuilder>::decodeMangledType(swift::Demangle::Node*) + 3260
8  libswiftCore.dylib             0x1b88f259c swift_getTypeByMangledNodeImpl(swift::MetadataRequest, swift::Demangle::Demangler&, swift::Demangle::Node*, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 500
9  libswiftCore.dylib             0x1b88f2330 swift::swift_getTypeByMangledNode(swift::MetadataRequest, swift::Demangle::Demangler&, swift::Demangle::Node*, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 540
10 libswiftCore.dylib             0x1b88f2820 swift_getTypeByMangledNameImpl(swift::MetadataRequest, llvm::StringRef, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 476
11 libswiftCore.dylib             0x1b88f050c swift::swift_getTypeByMangledName(swift::MetadataRequest, llvm::StringRef, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 540
12 libswiftCore.dylib             0x1b88e6a04 getSuperclassMetadata(swift::TargetClassMetadata<swift::InProcess>*, bool) + 272
13 libswiftCore.dylib             0x1b88e19fc _swift_initClassMetadataImpl(swift::TargetClassMetadata<swift::InProcess>*, swift::ClassLayoutFlags, unsigned long, swift::TypeLayout const* const*, unsigned long*, bool) + 80
14 myApp                          0x102c831ec type metadata completion function for ReplayBufferBase + 1976232
15 libswiftCore.dylib             0x1b88e8734 swift::MetadataCacheEntryBase<(anonymous namespace)::GenericCacheEntry, void const*>::doInitialization(swift::ConcurrencyControl&, swift::MetadataCompletionQueueEntry*, swift::MetadataRequest) + 268
16 libswiftCore.dylib             0x1b88dfc64 swift_getGenericMetadata + 1160
17 myApp                          0x102c842a4 type metadata accessor for ReplayBufferBase + 1980512
18 libswiftCore.dylib             0x1b88f5adc (anonymous namespace)::DecodedMetadataBuilder::createBoundGenericType(swift::TargetContextDescriptor<swift::InProcess> const*, llvm::ArrayRef<swift::TargetMetadata<swift::InProcess> const*>, swift::TargetMetadata<swift::InProcess> const*) const + 248
19 libswiftCore.dylib             0x1b88f4de0 swift::Demangle::TypeDecoder<(anonymous namespace)::DecodedMetadataBuilder>::decodeMangledType(swift::Demangle::Node*) + 3260
20 libswiftCore.dylib             0x1b88f259c swift_getTypeByMangledNodeImpl(swift::MetadataRequest, swift::Demangle::Demangler&, swift::Demangle::Node*, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 500
21 libswiftCore.dylib             0x1b88f2330 swift::swift_getTypeByMangledNode(swift::MetadataRequest, swift::Demangle::Demangler&, swift::Demangle::Node*, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const* (unsigned int, unsigned int)>, std::__1::function<swift::TargetWitnessTable<swift::InProcess> const* (swift::TargetMetadata<swift::InProcess> const*, unsigned int)>) + 540
22 libswiftCore.dylib             0x1b88f2820 swift_getTypeByMangledNameImpl(swift::MetadataRequest, llvm::StringRef, void const* const*, std::__1::function<swift::TargetMetadata<swift::InProcess> const [... cut ...]
42 libdispatch.dylib              0x1ab0b6610 _dispatch_call_block_and_release + 24
43 libdispatch.dylib              0x1ab0b7184 _dispatch_client_callout + 16
44 libdispatch.dylib              0x1ab063404 _dispatch_lane_serial_drain$VARIANT$mp + 608
45 libdispatch.dylib              0x1ab063df8 _dispatch_lane_invoke$VARIANT$mp + 420
46 libdispatch.dylib              0x1ab06d314 _dispatch_workloop_worker_thread + 588
47 libsystem_pthread.dylib        0x1ab106f88 _pthread_wqthread + 276
48 libsystem_pthread.dylib        0x1ab109ad4 start_wqthread + 8
```

mazenchami, 2019-11-06:
@Balake can you please post your `react-native info`?

Balake, 2019-11-06:
```
  React Native Environment Info:
    System:
      OS: macOS 10.15
      CPU: (8) x64 Intel(R) Core(TM) i7-4770HQ CPU @ 2.20GHz
      Memory: 4.12 GB / 16.00 GB
      Shell: 5.7.1 - /bin/zsh
    Binaries:
      Node: 10.15.3 - /usr/local/bin/node
      Yarn: 1.19.1 - /usr/local/bin/yarn
      npm: 6.11.3 - /usr/local/bin/npm
    SDKs:
      iOS SDK:
        Platforms: iOS 13.1, DriverKit 19.0, macOS 10.15, tvOS 13.0, watchOS 6.0
      Android SDK:
        API Levels: 24, 25, 26, 27, 28
        Build Tools: 27.0.3, 28.0.3
        System Images: android-28 | Google APIs Intel x86 Atom, android-29 | Google Play Intel x86 Atom
    IDEs:
      Android Studio: 3.4 AI-183.6156.11.34.5692245
      Xcode: 11.1/11A1027 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.8.6 => 16.8.6
      react-native: 0.59.10 => 0.59.10
    npmGlobalPackages:
      react-native-cli: 2.0.1
```

BinhTNT, 2019-11-07:
I get the same issue. Hope it would be fixed. Thanks

Balake, 2019-11-07:
I was able to fix this by downgrading to Xcode 11.1.

BinhTNT, 2019-11-07:
@Balake  I tried with Xcode 11.1, the crash didn't happen any more. Anyway, this is best solution currently. Thanks

ital0, 2019-11-11:
My five cents of contribution: even if you update to Xcode 11.2.1 GM Seed, the error still occurs.

gre, 2019-11-13:
Same issue, it crashes at boot of our app (in production build).

zhaiyjgithub, 2019-11-20:
I tried this way on Xcode 11.2.1 and works:  https://stackoverflow.com/questions/58595718/swift-initclassmetadataimpl-exc-bad-access-with-xcode-11-2-1

gre, 2019-11-20:
it seems that upgrading to React Native 0.61 (from 0.59) did the trick for us.

closed 2020-01-26 by stale[bot]

tdinckaya, 2021-12-05:
is there any solution?