dotintent/react-native-ble-plx #573: State gets Unsupported after refresh or JS crash

julioxavierr, 2019-11-13:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

After refreshing the app or having a JavaScript crash on iOS Bluetooth Adapter State should have the current state of BLE (same that it was if it wasn't changed by an external condition).

## Current Behavior

What is the current behavior?

Sometimes it comes as `Unsupported` even though it is `PoweredOn`, it just goes back to normal if the app is killed and opened again. Happens randomly.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Subscribe and log BLE state changes with `onStateChange`
1. Connect with a BLE device
1. Start monitoring data
1. Toggle RN commands and press Refresh
1. Notice that the state goes from Unknown to Unsupported and get stuck in this state

To have it happen it could be needed to run these steps more than once since it happens randomly.

### Context

```
System:
    OS: macOS 10.15
    CPU: (4) x64 Intel(R) Core(TM) i5-4308U CPU @ 2.80GHz
    Memory: 21.60 MB / 8.00 GB
    Shell: 3.2.57 - /bin/bash
  Binaries:
    Node: 10.13.0 - /var/folders/j0/mcc56qzn2934szgsqdwslc_40000gn/T/yarn--1573647094820-0.6661912929450491/node
    Yarn: 1.12.1 - /var/folders/j0/mcc56qzn2934szgsqdwslc_40000gn/T/yarn--1573647094820-0.6661912929450491/yarn
    npm: 6.5.0 - ~/.nvm/versions/node/v10.13.0/bin/npm
    Watchman: 4.9.0 - /usr/local/bin/watchman
  SDKs:
    iOS SDK:
      Platforms: iOS 13.1, DriverKit 19.0, macOS 10.15, tvOS 13.0, watchOS 6.0
    Android SDK:
      API Levels: 23, 24, 25, 26, 27, 28
      Build Tools: 23.0.1, 23.0.3, 25.0.2, 26.0.2, 27.0.3, 28.0.2, 28.0.3
      System Images: android-27 | Google Play Intel x86 Atom
      Android NDK: 16.1.4479499
  IDEs:
    Android Studio: 3.5 AI-191.8026.42.35.5791312
    Xcode: 11.1/11A1027 - /usr/bin/xcodebuild
  npmPackages:
    react: 16.9.0 => 16.9.0 
    react-native: 0.61.2 => 0.61.2
```

* Library version: 1.1.0
* Platform: iOS

Will keep adding more info as I troubleshoot this.

jukben, 2020-01-12:
Same issue. It has something to do with scanning the devices to soon after manager creation. 🤷 Haven't been able to figure it out yet, but for testing purposes I'm delaying BLE functions (scan) after BleManager creation by 1s.

Cierpliwy, 2020-01-31:
You should always wait for "PoweredOn" state before doing anything with a Bluetooth. You can use this call to monitor it: https://polidea.github.io/react-native-ble-plx/#blemanageronstatechange

ild-nucleusstudio, 2020-03-24:
@Cierpliwy I got the same issue.
After refreshing the app to update the new changes, the stage of BleManager changed to Unsupport.
I have to close the app and reopen again.

Any idea?

closed 2020-05-30 by stale[bot]

dan-lee, 2020-08-07:
Same problem here. It never transitions to `PoweredOn` in this case, only after hard restarting the app. Is there anything we can do?

wrsx, 2020-11-10:
We ran into this issue when initializing the BleManager in react state, causing react to re-initialize the BleManager on each render: 
`const [bleManager] = useState(new BleManager())`

We fixed it by moving the initialization outside the component. You could also use the callback form of useState:
`const [bleManager] = useState(() => new BleManager())`

ThomasStubbe, 2021-06-08:
Same issue