dotintent/react-native-ble-plx #574: Bluetooth printer just prints once, stops printing, and then prints duplicates

pdlm95, 2019-11-13:
## Prerequisites

- [X ] I am running the latest version
- [X ] I checked the documentation and found no answer
- [X ] I checked to make sure that this issue has not already been filed

## Expected Behavior

Print each time the button is pressed

## Current Behavior

The first time app sends print command, it prints. The second time (without any change), doesn't do anything. The next time, it prints duplicates (from the second and third time).

### Steps to Reproduce

1. Just call the method multiple times.

### Context

* Library version: 1.1.0
* Platform: Android.

As you can see, after scanning and connecting to the device, the print command is sent. After printing, I cancel the connection and then destroy the manager object in order to restart the object and be able to print again.

But this doesn't happen until the third time, when it prints duplicates. It does not show errors.

Maybe I got something wrong and I can't see it. Some help will be appreciated.

```js
export default class BluetoothSerial extends Component {
    constructor() {
        super()
        this.state = {info: "", values: {}}
    }

    info(message) {
        this.setState({info: message})
    }

   error(message) {
        this.setState({info: "ERROR: " + message})
    }

async scanAndConnect() {
        this.manager = new BleManager()
        this.manager.startDeviceScan(["000018f0-0000-1000-8000-00805f9b34fb"],
            null, async (error, device) => {
            this.info("Scanning...");
            console.log(device.name); //printer name

            if (error) {
                this.error(error.message);
                return
            }

            if (device.name === 'RG-MDP58B') {
                this.info("Connected to printer");
                this.manager.stopDeviceScan();

                await this.manager.connectToDevice(device.id)
                    .then(async (device) => {
                        await this.manager.discoverAllServicesAndCharacteristicsForDevice(device.id)
                        console.log('device id ', device.id)

                        await this.manager.writeCharacteristicWithoutResponseForDevice(
                             device.id,
                            "000018f0-0000-1000-8000-00805f9b34fb",
                            '00002af1-0000-1000-8000-00805f9b34fb', 
                            'cHJpbnRlciBjb25uZWN0ZWQ=')
                            .then((characteristic) => {
                                console.log('print value', characteristic.value)
                                this.info(characteristic.value);
                                this.manager.cancelDeviceConnection(device.id).then( () => {
                                    this.manager.destroy()
                                });
                            }).catch(err => console.log(err))
                    })
                    .catch((error) => {
                        this.error(error.message)
                    })
            }
        });
    }

    render() {
        return (
            <View>
                <Button title="Print" onPress={() => this.scanAndConnect()} />
                <Text>{this.state.info}</Text>
            </View>
        )
    }
}
```

pdlm95, 2019-11-14:
I feel so dumb, found the problem. It is not a problem of the library itself, It's the printer buffer.

I had to separate the code into multiple writeCharacteristicWithResponseForService with limited characters and now it works like a charm.

closed 2019-11-14 by pdlm95

FazilMuhammed, 2020-10-08:
@pdlm95 hello brother..i have also a bluetooth printer i need to print Image how its works.i can print text or letters successfully.please help me and waiting your response

pdlm95, 2020-10-08:
@FazilMuhammed I can't really tell how, never printed an image. You should check your printer requirements for images, maybe all you have to do its print a base64

FazilMuhammed, 2020-10-08:
@pdlm95  i converted the images into base64 while printing something wrong letters coming..

devnullpointer, 2020-11-13:
@pdlm95 if i may ask, how did you get your printers UUID? I'm trying to connect to a bluetooth label printer but I only have its BT Mac. Thanks.

pdlm95, 2020-11-16:
@devnullpointer you should first try looking for information in the manufacturer's manual. If there's nothing, you could try installing an app that tells you the characteristics of the device. You could check nRF connect

mayniii, 2020-12-07:
@pdlm95 can u provide the example for print text? how u convert printer command to base64?

pdlm95, 2020-12-07:
@mayniii This is how I made it. I provide the text that I want to print in an array, where every line I wish to print its an array item.

```js
import { Buffer } from 'buffer';

const SERVICE_UUID = '000018f0-0000-1000-8000-00805f9b34fb';    //primary service (printers)
const CHAR_UUID = '00002af1-0000-1000-8000-00805f9b34fb';       //characteristic (write)

async print(text: array){
    await this.scanAndConnect().then(async (res: any) => {
            for (let item of text) {
                let encryptedCredentials = new Buffer(item).toString("base64");

                promises.push(res.Device.writeCharacteristicWithResponseForService(
                    SERVICE_UUID, CHAR_UUID, encryptedCredentials));
            }
            await Promise.all(promises).then(() => {
                this.manager.destroy()
            })
        }).catch(err => {return});
}

AyishmAzeem, 2022-01-18:
@pdlm95 can you guide me about receipt printing?