dotintent/react-native-ble-plx #575: iOS 13 Bluetooth permission prompted immediately after BleManager was only initialized

BohdanIvanov, 2019-11-15:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Hello everyone. I'm facing a bit of a problem with iOS 13 version. After the app was booted and BleManager was just initialized a user gets immediately prompted that he needs to allow/decline permission to use the Bluetooth in the application, but the app was only initialized and there were no Bluetooth actions yet. I'm curious, is there a way to ask permission later, when Bluetooth will be actually used?

Thank you in advance

jbrun, 2019-12-21:
I also have the same problem, have you found a solution?

BohdanIvanov, 2019-12-22:
Nope :(

Cierpliwy, 2020-01-31:
You can always initialize BleManager later. There is no API to control a moment when iOS asks for permissions.

closed 2020-04-07 by stale[bot]