dotintent/react-native-ble-plx #579: Missing Purpose String in Info.plist

schwamic, 2019-11-26:
The app's Info.plist file should contain a **NSBluetoothPeripheralUsageDescription** key with a user-facing purpose string explaining clearly and completely why your app needs the data. Starting Spring 2019, all apps submitted to the App Store that access user data are required to include a purpose string.

Maybe the **NSBluetoothPeripheralUsageDescription** should be added to the documentation?

Cierpliwy, 2020-01-31:
We definitely need to add it!

dididy, 2020-12-21:
I had the same problem. And I found out that this issue was the cause. Could I add it on Document? I think it's a worthwhile task to make sure it's not confusing.

mciechanowicz, 2022-01-04:
Already fixed here: https://github.com/expo/config-plugins/pull/13

closed 2022-01-04 by mciechanowicz