dotintent/react-native-ble-plx #581: Sending long data to BLE Device

swapniliziss, 2019-11-27:
I have to send big data (60 bytes)Android, for that I am increasing MTU by using "requestMTUForDevice", but it not increase. Could you help me how i send long data to BLE Device?

pstanton, 2019-12-16:
chunk your data

Cierpliwy, 2020-01-31:
Requesting MTU doesn't guarantee to change it. If it returns for example 23, you should as @pstanton said chunk your data. Chunking data is necessary only when using notifications and writes without response. Otherwise, you should be able to write values up to 512 bytes and the LE layer should chunk it for you.

rfbezerra, 2020-02-03:
I achivied this using the *RxBleConnection.LongWriteOperationBuilder*.
Is there any plan to have an "safeLongWriteCharacteristicForDevice" or anything like that officially?

closed 2020-04-10 by stale[bot]