dotintent/react-native-ble-plx #586: Experiencing Freeze while monitor

ScreamZ, 2019-12-10:
# Introduction

I'm currently exchanging data with a custom made Bluetooth device and implementing a custom protocol. For that, I work with the device builder and we made some tries using `nRF Connect` an app on Android to test Bluetooth exchange, no issue detected with that app and everything works well. 

The device is sending a chunk of data through 80 Bytes each time using an RX characteristic, from there I parse the binary data into an exploitable one. I can successfully monitor with through the indicate, but after some frame (it's unclear), the JS app is getting frozen and I'm not able to interact anymore (using touch or anything). Same for JS logs.

The only thing I have to do is kill the app and start over.

## Expected Behavior

I receive periodic indication, without any freezing on my app. I can decode properly.

## Current Behavior

I get some frame, then the app completely freezes on both IOS and ANDROID (same issue).

## Context

* Library version: `1.1.0`
* Platform: `Android & iOS`.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```JSON
 "dependencies": {
    "@react-native-community/masked-view": "^0.1.5",
    "@react-native-firebase/app": "^6.0.4",
    "@react-native-firebase/crashlytics": "^6.0.4",
    "@react-native-firebase/remote-config": "^6.0.4",
    "@react-navigation/core": "^5.0.0-alpha.24",
    "@react-navigation/native": "^5.0.0-alpha.16",
    "@react-navigation/stack": "^5.0.0-alpha.37",
    "await-to-js": "^2.1.1",
    "bin-grammar": "^3.3.0",
    "buffer": "^5.4.3",
    "date-fns": "^2.8.1",
    "i18next": "^19.0.1",
    "lodash": "^4.17.15",
    "react": "^16.9.0",
    "react-error-boundary": "^1.2.5",
    "react-i18next": "^11.2.2",
    "react-native": "^0.61.5",
    "react-native-ble-plx": "^1.1.0",
    "react-native-gesture-handler": "^1.5.1",
    "react-native-localize": "^1.3.1",
    "react-native-reanimated": "^1.4.0",
    "react-native-safe-area-context": "^0.6.0",
    "react-native-screens": "^1.0.0-alpha.23",
    "react-native-typography": "^1.4.1",
    "styled-components": "^4.4.1"
  },
  "devDependencies": {
    "@babel/core": "^7.7.4",
    "@babel/runtime": "^7.7.4",
    "@types/binary-parser": "^1.3.1",
    "@types/lodash": "^4.14.149",
    "@types/react-native": "^0.60.22",
    "@types/styled-components": "^4.4.0",
    "@welldone-software/why-did-you-render": "^3.3.9",
    "babel-jest": "^24.9.0",
    "babel-plugin-module-resolver": "^3.2.0",
    "husky": "^3.1.0",
    "jest": "^24.9.0",
    "lint-staged": "^9.4.3",
    "metro-react-native-babel-preset": "^0.57.0",
    "prettier": "^1.19.1",
    "reactotron-react-native": "^3.0.0",
    "tslint": "^5.20.1",
    "tslint-config-prettier": "^1.18.0",
    "tslint-plugin-prettier": "^2.0.1",
    "tslint-react": "^4.1.0",
    "tslint-react-hooks": "^2.2.1",
    "typescript": "^3.7.2",
    "typescript-styled-plugin": "^0.15.0"
  }
```

```tsx
import React, { useEffect, useState } from "react";
import { View, Platform, PermissionsAndroid, Button } from "react-native";
import _ from "lodash";
import { getFormatedDecodedFrame } from "@modules/bluetooth";
import { useBleManager } from "@modules/bluetooth/hooks/useBleManagerContext";
import { Device, Subscription, BleError, Characteristic } from "react-native-ble-plx";
import { Buffer } from "buffer";

export const WelcomeScreen: React.FC = () => {
  const BleManager = useBleManager();
  const [rxListener, setRxListener] = useState<Subscription>();
  const [detectedDevices, setDevices] = useState<Device[]>([]);

  useEffect(() => {
    if (Platform.OS === "android" && Platform.Version >= 23) {
      PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION);
    }

    const subscription = BleManager.onStateChange((state: string) => {
      if (state === "PoweredOn") {
        // Stop state change listener.
        subscription.remove();

        // Start scanning devices
        BleManager.startDeviceScan(null, null, (error, device) => {
          if (error) {
            // TODO: Do something
            return;
          }

          // We ignore devices without proper
          if (!device || !device.name || device.localName !== "GEN_V4") {
            return;
          }

          // Stop scan
          BleManager.stopDeviceScan();

          // Update devices list
          setDevices([device]);
        });
      }
    }, true);
  }, []);

  return (
    <View style={{ flex: 1, justifyContent: "center" }}>
      {_(detectedDevices)
        .uniqBy("id")
        .map(device => (
          <React.Fragment key={device.id}>
            <Button
              title={device.localName || "Missing name ?"}
              // tslint:disable-next-line: jsx-no-lambda
              onPress={async () => {
                // TODO: Should I do this ?
                // if (await device.isConnected) {
                //   await device.cancelConnection();
                // }

                await device.connect({ autoConnect: true, requestMTU: 80 });
                console.log(`Connected to device: ${device.id}, discovering services and caracteristics…`);

                await device.discoverAllServicesAndCharacteristics();
                console.log(`Services and Characteristics discovered!`);

                const chararacteristics = await device.characteristicsForService("09912756-7b32-4629-aeb1-b309d9a338ae");
                console.log(`Found characteristics for service: 09912756-7b32-4629-aeb1-b309d9a338ae!`);

                const RX = chararacteristics.find(char => char.uuid === "ef785c24-22bb-463d-b651-0b7445ba091c");

                if (RX) {
                  console.log(`Found RX characteristics (ef785c24-22bb-463d-b651-0b7445ba091c)!`);
                  const subscription = RX.monitor(onIndicate);

                  setRxListener(subscription);
                }
              }}
            />
            <Button
              title="Disconnect"
              onPress={() => {
                rxListener?.remove();
                device.cancelConnection();
              }}
            />
          </React.Fragment>
        ))
        .value()}
    </View>
  );
};

function onIndicate(err: BleError | null, readCharacteristic: Characteristic | null): void {
  try {
    if (
      err ||
      !readCharacteristic ||
      !readCharacteristic.value ||
      readCharacteristic?.value ===
        "AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="
    ) {
      console.log("Error or no characteristic", err);
      return;
    }

    console.log("(BASE64) New Frame received:", readCharacteristic.value, " Creating Buffer…");
    const buffer = Buffer.from(readCharacteristic?.value, "base64");
    console.log("(HEX) Given Frame converted:", buffer.toString("hex"));

    // Convert frame using npm package `import { uint16, parse, selector, uint8 } from "bin-grammar";`
    const frame = getFormatedDecodedFrame(buffer);
    console.log("(OBJECT) Decoded frame as JSON", frame);
  } catch (e) {
    console.log("OnIndicate error: ", e);
  }
}
```

```tsx
import React, { createContext, useContext, useRef } from "react";
import { BleManager } from "react-native-ble-plx";

// This is used as a singleton here as we are not providing anything
const BleManagerContext = createContext<BleManager>(new BleManager());

/**
 * Access to the manager anywhere in the app.
 *
 * @export
 * @returns {AppContext}
 */
export function useBleManager(): BleManager {
  return useContext(BleManagerContext);
}

export const BleManagerProvider: React.FC = React.memo(props => {
  const managerRef = useRef<BleManager>(new BleManager({}));

  return <BleManagerContext.Provider value={managerRef.current!}>{props.children}</BleManagerContext.Provider>;
});
```

Do you have any idea why this could happen?

It looks like it freezes less frequently when I'm not decoding the frame using `binary-grammar`  so I was asking myself if there is something that waits for a response and my code might be too long to run?

I didn't turn the system logs on, will try tomorrow… But maybe you already have some ideas ?

Thanks a lot,
Regards,

Andréas

ScreamZ, 2019-12-11:
Looks like it was related to my parsing library…

closed 2019-12-11 by ScreamZ

neemiasbj, 2022-03-15:
Hi @ScreamZ,

Did you use expo?
and did your app works after build release?