dotintent/react-native-ble-plx #588: Support autolinking > 0.60

ScreamZ, 2019-12-12:
Hi,

Did you saw that and is it planned ?
```
error React Native CLI uses autolinking for native dependencies, but the following modules are linked manually: 
  - react-native-ble-plx (to unlink run: "react-native unlink react-native-ble-plx")
This is likely happening when upgrading React Native from below 0.60 to 0.60 or above. Going forward, you can unlink this dependency via "react-native unlink <dependency>" and it will be included in your app automatically. If a library isn't compatible with autolinking, disregard this message and notify the library maintainers.
Read more about autolinking: https://github.com/react-native-community/cli/blob/master/docs/autolinking.md
```

Regards

Cierpliwy, 2020-01-31:
There are plans to release version 2.0, which will target RN 0.60+ and use autolinking feature as far as I know. There are no API changes planned for 2.0.

thibaultdewit, 2020-03-14:
So it is fine to disregard the instructions regarding unlinking this package at the moment?

nsantacruz, 2020-03-14:
Yes. Manual linking still works currently.

RANGELJ, 2020-03-24:
So autolinking does not work yet?

closed 2020-05-30 by stale[bot]

azuxx, 2020-07-30:
any updates about this? Warning still exists so I think I have to keep it manual still.

dariuszseweryn, 2020-07-30:
There is version 2.0.1. Current linking process is described in readme.

azuxx, 2020-07-30:
> There is version 2.0.1. Current linking process is described in readme.

@dariuszseweryn Thank you, I have already have  "react-native-ble-plx": "^2.0.1" in my package.json but then I followed the wrong guide here:  
[image]

there is no mention about autolinking here.

dariuszseweryn, 2020-07-30:
[Automatic linking is the first two steps](https://reactnative.dev/docs/linking-libraries-ios#automatic-linking). There are additional steps related to BLE feature and the way this library is implemented on each platform and are obligatory.

azuxx, 2020-07-30:
> [Automatic linking is the first two steps](https://reactnative.dev/docs/linking-libraries-ios#automatic-linking). There are additional steps related to BLE feature and the way this library is implemented on each platform and are obligatory.

@dariuszseweryn  I have updated to RN 0.63.2 and I don't have those stuff in my xCode project. Anyhow my podfile looks like this and your library at the moment is working properly, it was just about that warning I was having when running npx react-native run-ios.

```
require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'
platform :ios, '10.0'

target 'XXXReactNative' do

  config = use_native_modules!
  use_react_native!(:path => config["reactNativePath"])
  pod 'react-native-ble-plx', :path => '../node_modules/react-native-ble-plx'

  #target 'XXXReactNativeTests' do
    #inherit! :complete
    # Pods for testing
  #end

  # Enables Flipper.
  #
  # Note that if you have use_frameworks! enabled, Flipper will not work and
  # you should disable these next few lines.
  use_flipper!
  post_install do |installer|
    flipper_post_install(installer)
  end
end

#target 'XXXReactNative-tvOS' do
  # Pods for XXXReactNative-tvOS

  #target 'XXXReactNative-tvOSTests' do
  #  inherit! :search_paths
    # Pods for testing
  #end
#end

```