dotintent/react-native-ble-plx #589: On IOS, localname is missing sometimes

ScreamZ, 2019-12-13:
On IOS,

1.11, IOS13

While scanning I can access device name, but the result of the promise I don't.

```ts
const connectedDevice = await device.connect({ requestMTU: 80 })

// device.name === Good name
// connectedDevice.name === null
```

I also miss a lot of data, a seen on my screenshot.

[image]

ScreamZ, 2020-01-13:
Also it looks like it's not correct to call connect that way

```tsx
  async function connectToDevice(deviceId: string, saveForAutoConnect = false) {
    try {
      const bluetoothEnabled = await ensureBluetoothEnabled();

      if (!bluetoothEnabled) {
        return false;
      }

      // Disconnect eventually
      await bleManager.current
        .cancelDeviceConnection(deviceId)
        .catch(() => undefined);

      // Attempt connection
      const [errConnect, device] = await to(
        bleManager.current.connectToDevice(deviceId, {
          timeout: 15000,
        }),
      );

      // Sometimes happens on non-genesis device when not bound
      if (errConnect || !device) {
        Alert.alert(
          "Erreur de connexion",
          "[errConnect]: " + errConnect.message + "|" + errConnect.reason,
        );
        // TODO: Crashlytics to handle case that are not listed
        return false;
      }

      // Connection successful

      // Discover services and characteristics
      const [errDiscover] = await to(
        device.discoverAllServicesAndCharacteristics(),
      );

      console.log("Discovery finished");
      if (errDiscover) {
        Alert.alert(
          "Erreur de connexion",

          "[errDiscover]: " + errDiscover.message,
        );
        // Dispose connection
        await disconnectedFromDevice();

        return false;
      }

      // Ensure we are on a genesis device
      const [errGetServiceChars, chararacteristics] = await to(
        device.characteristicsForService(BleServices.EPOOL),
      );
      if (
        errGetServiceChars ||
        !chararacteristics ||
        chararacteristics.length === 0
      ) {
        Alert.alert(
          "Erreur de connexion",

          "[errGetServiceChars]: " +
            JSON.stringify({
              chararacteristics,
              errGetServiceChars:
                errGetServiceChars.message + "|" + errGetServiceChars.reason,
            }),
        );
        console.log(device);

        // Dispose connection
        await disconnectedFromDevice();

        return false;
      }

      // Bind
      setCurrentDevice(device);

      // Save device
      if (saveForAutoConnect) {
        await setSavedDevices(
          Array.isArray(savedDevices)
            ? savedDevices.concat([device.id])
            : [device.id],
        );
      }

      return true;
    } catch (e) {
      Alert.alert(
        "Une erreur est survenue",
        "[connectToDevice fn]: " + e.message + "|" + e.reason,
      );
      return false;
    }
  }
```

As device instance as https://polidea.github.io/react-native-ble-plx/#device says Device instance which can be retrieved only by calling bleManager.startDeviceScan().
I'm having many issues achieving similar behavior between IOS and android
I'll give a small brief about what I need to do and what I'm currently doing

I have a custom made IOT device, which speaks through BLE (this device is built by my customer, so it has an hand on the configuration).

Now I need to do some things,

1. I scan for devices and on connect, I save in AsyncStorage the device ID.
2. On various case, I try to connect using the above function using the saved deviceId
3. The thing is sometimes I get ejected by the code that says I'm not connected, whereas I'm connected.
4. As you can see I'm always discovering… The thing is there are some issues on whether the device is connected or not. As far as I understand, some informations are advertising one, which are reported by a scan, it that case how can I get those when I connect directly from device ?

ScreamZ, 2020-01-13:
Here is my log also: 

```
2020-01-13 11:24:04.284492+0100 Epool[618:205060] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a60740> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-01-13 11:24:04.300927+0100 Epool[618:205060] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a58c80> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-01-13 11:24:04.306087+0100 Epool[618:205059] [CoreBluetooth] XPC connection invalid
2020-01-13 11:24:04.312931+0100 Epool[618:205060] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a59280> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
[RxBLEKit|DEBG|10:24:04.316]: CentralManager(10748269184) didUpdateState(state: poweredOn)
2020-01-13 11:24:04.317463+0100 Epool[618:205051] [CoreBluetooth] XPC connection invalid
2020-01-13 11:24:04.385680+0100 Epool[618:205060] 6.8.1 - [Firebase/RemoteConfig][I-RCN000069] Most recently fetched config is already activated.
2020-01-13 11:24:04.433748+0100 Epool[618:205060] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a74f80> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
[RxBLEKit|DEBG|10:24:04.437]: CentralManager(10748383104) didUpdateState(state: poweredOn)
2020-01-13 11:24:04.439205+0100 Epool[618:205058] [CoreBluetooth] XPC connection invalid
2020-01-13 11:24:04.456174+0100 Epool[618:205058] [] nw_socket_handle_socket_event [C8:1] Socket SO_ERROR [61: Connection refused]
2020-01-13 11:24:04.458395+0100 Epool[618:205060] [] nw_connection_get_connected_socket [C8] Client called nw_connection_get_connected_socket on unconnected nw_connection
2020-01-13 11:24:04.458606+0100 Epool[618:205060] TCP Conn 0x282734780 Failed : error 0:61 [61]
2020-01-13 11:24:04.478646+0100 Epool[618:205051] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a59700> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
[RxBLEKit|DEBG|10:24:04.483]: CentralManager(10748270336) didUpdateState(state: poweredOn)
2020-01-13 11:24:04.481926+0100 Epool[618:205059] [CoreBluetooth] XPC connection invalid
2020-01-13 11:24:04.489454+0100 Epool[618:205058] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a592c0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-01-13 11:24:04.500964+0100 Epool[618:205058] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a741c0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-01-13 11:24:04.506033+0100 Epool[618:205050] [CoreBluetooth] XPC connection invalid
2020-01-13 11:24:04.506072+0100 Epool[618:205058] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a592c0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-01-13 11:24:04.511428+0100 Epool[618:205051] [CoreBluetooth] XPC connection invalid
[RxBLEKit|DEBG|10:24:04.511]: CentralManager(10748269248) didUpdateState(state: poweredOn)
2020-01-13 11:24:04.511541+0100 Epool[618:205060] [CoreBluetooth] XPC connection invalid
[RxBLEKit|DEBG|10:24:04.726]: CentralManager(10748269248) retrievePeripherals(
withIdentifiers: [19B285D8-0293-C26A-6D5E-E776C000EF13])
[RxBLEKit|DEBG|10:24:04.732]: CentralManager(10748269248) connect(
peripheral: Peripheral(uuid: 19B285D8-0293-C26A-6D5E-E776C000EF13, name: Optional("BLE Device")),
options: nil)
2020-01-13 11:24:04.760 [info][tid:main][RNGestureHandlerManager.m:136] [GESTURE HANDLER] Initialize gesture handler for root view <RCTRootContentView: 0x10223b3c0; reactTag: 11; frame = (0 0; 375 667); gestureRecognizers = <NSArray: 0x281c47630>; layer = <CALayer: 0x28125c4c0>>
[RxBLEKit|DEBG|10:24:05.132]: Peripheral(uuid: 19B285D8-0293-C26A-6D5E-E776C000EF13, name: Optional("BLE Device")) discoverServices(
serviceUUIDs: nil)
[RxBLEKit|DEBG|10:24:05.419]: Peripheral(uuid: 19B285D8-0293-C26A-6D5E-E776C000EF13, name: Optional("BLE Device")) didDiscoverServices(services
: Optional("[Service(uuid: 09912756-7B32-4629-AEB1-B309D9A338AE, id: 10748274240), Service(uuid: 0E391B4A-D4A4-4184-A2E2-A1BC3A34C3E5, id: 10748280000)]"),
error: nil)
[RxBLEKit|DEBG|10:24:05.436]: Peripheral(uuid: 19B285D8-0293-C26A-6D5E-E776C000EF13, name: Optional("BLE Device")) discoverCharacteristics(
characteristicUUIDs: nil,
for: Service(uuid: 09912756-7B32-4629-AEB1-B309D9A338AE, id: 10748274240))
[RxBLEKit|DEBG|10:24:05.439]: Peripheral(uuid: 19B285D8-0293-C26A-6D5E-E776C000EF13, name: Optional("BLE Device")) discoverCharacteristics(
characteristicUUIDs: nil,
for: Service(uuid: 0E391B4A-D4A4-4184-A2E2-A1BC3A34C3E5, id: 10748280000))
[RxBLEKit|DEBG|10:24:05.483]: Peripheral(uuid: 19B285D8-0293-C26A-6D5E-E776C000EF13, name: Optional("BLE Device")) didDiscoverCharacteristicsFor(for:
Service(uuid: 09912756-7B32-4629-AEB1-B309D9A338AE, id: 10748274240), characteristics:
Optional("[Characteristic(uuid: 4D32C5E5-2BB5-45C6-8C89-6F59BB3930D2, id: 10794117632), Characteristic(uuid: EF785C24-22BB-463D-B651-0B7445BA091C, id: 10794121664), Characteristic(uuid: 69C2A4AE-6A00-45F1-979F-F58A070C7060, id: 10794110 [... cut ...]
I'm getting a lot of `[CoreBluetooth] API MISUSE: <CBCentralManager: 0x280a42600> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
[RxBLEKit|DEBG|10:24:06.158]: CentralManager(10748175872) didUpdateState(state: poweredOn)`

Also you can see that the device is named `BLE Device`.

When I  called `device.characteristicsForService(serviceId)` i get Device X19B…F13 is not connected

@Cierpliwy @piotrdubiel @dariuszseweryn maybe you could help me on that guys ?

Thanks a lot

Cierpliwy, 2020-01-31:
Did you checked https://polidea.github.io/react-native-ble-plx/#blemanagerconnecteddevices and used it?

ScreamZ, 2020-02-01:
Not yet, will try

Harneetk, 2020-04-05:
Still not resolved. Even we are facing same issue while scanning. Complete advertising data is not there. Pl confirm if there is any resolution for same.

diogoviannaaraujo, 2020-07-02:
Having the same problem here, after MTU request the received object from the promise is almost empty, except from the new MTU. using Manager,connectedDevices after mtu request also brings empty data like the one in the picture.

Also could not confirm if the MTU was applied, i’ve been having slow reads, about 1 second to read about 250 bytes, my search about why its so slow made me notice this bug, I wonder if they are related.

I can reproduce easily in more that one device and have access to BLE Server logs too, if you need more detailed information about this, i can provide.

mciechanowicz, 2022-01-07:
please check this one: https://github.com/dotintent/react-native-ble-plx/issues/939

closing for now. if it still will be a problem please reopen.

closed 2022-01-07 by mciechanowicz