dotintent/react-native-ble-plx #590: node module cannot be null. please help thank you!

isa3bel, 2019-12-16:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior
app can run
Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?
[image]

### Context

* Library version: X.Y.Z
* Platform: iOS.
* Contents of the `package.json` file:
```{
  "name": "MountLock",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "firebase": "^7.2.2",
    "js-base64": "^2.5.1",
    "js-sha3": "^0.8.0",
    "react": "^16.11.0",
    "react-native": "0.61.5",
    "react-native-ble-plx": "^1.1.0",
    "react-native-gesture-handler": "^1.4.1",
    "react-navigation": "^4.0.10",
    "react-navigation-stack": "^1.10.3"
  },
  "devDependencies": {
    "@babel/core": "^7.6.4",
    "@babel/runtime": "^7.6.3",
    "@react-native-community/eslint-config": "^0.0.5",
    "babel-jest": "^24.9.0",
    "eslint": "^6.6.0",
    "jest": "^24.9.0",
    "metro-react-native-babel-preset": "^0.57.0",
    "react-test-renderer": "16.12.0"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

dariuszseweryn, 2019-12-16:
Hello,
It seems that you have not integrated the library correctly. Try going through the integration process once again (check readme)

isa3bel, 2019-12-16:
Thanks for the comment! What made you think I did't integrate the library correctly? this will help me to figure out the problem. thanks again!

dariuszseweryn, 2019-12-16:
There are [other similar issues here that are already closed](https://github.com/Polidea/react-native-ble-plx/issues?utf8=%E2%9C%93&q=is%3Aissue+native+module+cannot+be+null)

isa3bel, 2019-12-17:
I’m cloning another project that uses this ble library however should I follow the Installation steps manually after cloning the project?

dariuszseweryn, 2019-12-17:
It depends if the project was correctly set up. I expect that [this](https://github.com/Cierpliwy/SensorTag) would only need `npm install` and then run the project

isa3bel, 2019-12-17:
ok thank you! 
[image] I get this error now which isn't the same one as before which is good. I think this will be easier to fix

dariuszseweryn, 2019-12-19:
Closing then

closed 2019-12-19 by dariuszseweryn