dotintent/react-native-ble-plx #592: Is there a way to reduce scanning time?

martimalek, 2019-12-19:
Hello guys! I've been using this library for a few weeks and I can say that it works pretty well. I can scan, connect and do lots of stuff with ble devices with simplicity, so I am very pleased with it. 

Right now I'm working on a project which connects from a RN app to a Raspberry Pi through BLE. It has one service which contains just one characteristic, pretty simple right?

Well... my only issue with this library right now is the time it takes to find my device. Which seems pretty random (it can take as little as .3seconds to as much as 15seconds, normally it takes 4-5seconds) and I don't know what else I can do to reduce it.

This is the relevant code:
```
bleManager.current.startDeviceScan(UUIDS, null, async (error, device) => {
            if (error) throw Error(error);
            if (device && device.id === MAC) {
                bleManager.current.stopDeviceScan(); // from startDeviceScan to here 4seconds have passed T.T
                const connectedDevice = await device.connect();
                const usefulDevice = await connectedDevice.discoverAllServicesAndCharacteristics();
                setBleDevice(usefulDevice);
                usefulDevice.onDisconnected((err, dev) => console.log('Device disconnected', err, dev));
            }
        });
```
In this example UUIDS is an array of ids like
``` 
['myUUID']
```

I'm already passing a specific UUID to scan for, so this should reduce the time it takes to scan, right?
Also when I say *the time it takes to find my device* I mean just scan it, not connect to it or discover it's services. This means that the user will wait 4-5seconds to scan for the device + the time to connect to it + discover it's services and characteristics, which could go up to **6-20seconds**. **This is hell for the user**

By the way, my raspberry's advertising interval is .3seconds so this shouldn't be an issue of the peripheral. Also I've tried scanning for the Raspberry from Nordic apps (nRF Connect / nRF Toolbox) and they findmy device in less than 1 second without exception.

So I'm clearly doing something wrong here but can't find anything in the docs which helps me. I'm thinking that maybe the issue has something to do with calling startDeviceScan from a component and not from the root file (not sure at all). Also I'm assigning bleManager to a react ref, but I don't think that this should make a difference in scanning time. 

By the way I'm using RN 0.61.4 and react-native-ble-plx 1.1.0.
If anyone has any advice, comment, suggestion anything is much appreciated.

naresh-rently, 2019-12-26:
@martimalek Could you try passing { scanMode: ScanMode.LowLatency } to startDeviceScan instead of null.

martimalek, 2020-01-02:
@naresh-rently Thank you very much for your help. This change does indeed seem to improve the speed by a little bit. Right now the app never takes more than 5 seconds to scan + connect + discover characteristics of a device (**which is a huge time decrease respect the 6-20seconds from earlier**), the average time it takes is 3-4seconds.
I'm also using a foreground service so that when the user opens the app it already has connected to a device if in range.

Even though this is a huge upgrade respect last version I'm sure that there is still room for improvement. For now though, I'm going to close this issue

closed 2020-01-02 by martimalek