dotintent/react-native-ble-plx #594: readCharacteristicForService does not work

tknuts, 2019-12-29:
After device.connect() and even after discoverAllServicesAndCharacteristics(), readCharacteristicForService does not work. But getting charactaristics by digging in does?

The first does not work:

```
async readDoorCharacteristic() {
    this.openDoorCharacteristic = await this.device.readCharacteristicForService("947cd7c8-e2cb-11e9-81b4-2a2ae2dbcce4", "947cda5c-e2cb-11e9-81b4-2a2ae2dbcce4");
}
```

This one does:

```
  async readCharacteristics() {
    this.openDoorCharacteristic = null
    
    if (this.connected) {
      const services = await this.device.services()
      for (let i = 0; i < services.length; i++) {
        const service = services[i];
        this.log.push('Service uuid: ' + service.uuid)
        const characteristics = await service.characteristics()
        const openDoorCharacteristic = characteristics.find(c => c.uuid === "947cda5c-e2cb-11e9-81b4-2a2ae2dbcce4")
        if (openDoorCharacteristic) {
          this.openDoorCharacteristic = openDoorCharacteristic
        }
      }
    }
  }
```
But I would prefer not itterating through services and caharactaristics, when they are known to me...

Any ideas why this doesn't work?

ps. The serviceUuid and CharactaristicUuid are not the same, but close :-)

Thanks.. Regards, Thomas

jimji1005, 2020-01-01:
discover all service and characteristic after connecting, then write to it or read from it. until you run discovery, they can not be accessed. 
https://github.com/Polidea/react-native-ble-plx/wiki/Device-Service-Discovery

Cierpliwy, 2020-01-31:
`device.readCharacteristicForService` and `characteristic.read()` are using similiar function calls under the hood. As @jimji1005 mentioned, make sure to discover everything and double check UUIDs.

closed 2020-04-07 by stale[bot]