dotintent/react-native-ble-plx #595: SetupNotifications (monitorCharacteristicForService)

kot7kzz, 2020-01-07:
Hello, my first post on github so i'm sorry if something is wrong with formatting/rules

I'm designing an app that connects via bluetooth to an heart rate sensor and with this library i successfully connect and read the values from it.

My issue here is that this one service is notify only (no permissions to read or write) and the only function from the library that i managed to use was monitorCharacteristicForService.

Everything was good until the program got complex. The issue here is that i can only get the values INSIDE the monitor function and take it outside of it with setState (which causes automatic render etc etc).

Any suggestions on how i can have full control on that "buf[1]" variable on the image? all the values i get are correct, i just want to not use setState or manage to get values from the characteristic outside the function so i have more room to work with.

[image]

Once again sorry for the noob question since it's not really a library issue but more like i don't have the skills to completely understand the documentation of monitorCharacteristicForService.

Thanks in advance for all the help you can provide!

PS: i tried using redux to copy the state instead of changing it but without success

Cierpliwy, 2020-01-31:
What I would suggest it to convert `monitorCharacteristicForService` into a promise, which is buffered and fetch values after a certain while. Example test how it could work:

```ts
export interface Subscription {
  remove(): void;
}

// Fake device for simulation...
class Device {
  monitorCharacteristicForDevice(
    listener: (value: number | null, error: Error | null) => void
  ): Subscription {
    let value = 0;
    let interval = setInterval(() => {
      if (value < 20) {
        listener(value, null);
        value++;
      } else {
        listener(null, new Error("Disconnected"));
        clearInterval(interval);
      }
    }, 200);
    return {
      remove: () => {
        clearInterval(interval);
      }
    };
  }
}

// Function used to construct an emitter.
type Emitter<T, E extends Error> = (
  emit: (value: T | null, error: E | null) => void
) => () => void;

// Our API to convert any emitter funtion into a buffer
class EmitterBuffer<T, E extends Error> {
  private values: Array<T>;
  private error: E | null;
  private subscription: (() => void) | null;
  private promises: Array<{
    resolve: (value: Array<T>) => void;
    reject: (error: E) => void;
  }>;

  constructor(emitter: Emitter<T, E>) {
    this.values = [];
    this.error = null;
    this.promises = [];
    this.subscription = emitter((value, error) => {
      if (value != null) {
        this.resolve(value);
      }
      if (error != null) {
        this.reject(error);
      }
    });
  }

  resolve(value: T) {
    let resolvedAny = false;
    this.values.push(value);
    this.promises.forEach(promise => {
      promise.resolve(this.values);
      resolvedAny = true;
    });
    if (resolvedAny) {
      this.values = [];
      this.promises = [];
    }
  }

  reject(error: E) {
    let rejectedAny = false;
    this.error = error;
    this.promises.forEach(promise => {
      promise.reject(error);
      rejectedAny = true;
    });
    if (rejectedAny) {
      this.promises = [];
    }
  }

  fetchValues(): Array<T> {
    if (this.error != null) {
      throw this.error;
    }

    const values = this.values.splice(0, this.values.length);
    return values;
  }

  waitAndFetchValues(): Promise<Array<T>> {
    if (this.error != null) {
      return Promise.reject(this.error);
    }
    return new Promise((resolve, reject) => {
      this.promises.push({
        resolve,
        reject
      });
    });
  }

  close() {
    if (this.subscription != null) {
      this.subscription();
      this.subscription = null;
    } else {
      throw new Error("EmitterBuffer already closed!");
    }
  }
}

// async/await delay
function delay(timeout: number): Promise<void> {
  return new Promise(resolve => {
    setTimeout(resolve, timeout);
  });
}

// Testing....
const device = new Device();
const monitorBuffer = new EmitterBuffer(emitter => {
  const subscription = device.monitorCharacteristicForDevice((value, error) => {
    if (value != null) {
      emitter(value, null);
    }
    if (error != null) {
      emitter(null, error);
    }
  });
  return () => {
    subscription.remove();
  };
});

const doTests = async () => {
  try {
    for (;;) {
      // This may result in empty arrays:
      // await delay(Math.random() * 1000);
      // const values = monitorBuffer.fetchValues();
      // console.log(values);

      // This will not result in empty arrays, because we are waiting for values.
      await delay(Math.random() * 1000);
      const values = await monitorBuffer.waitAndFetchValues();
      console.log(values);
    }
  } catch (error) {
    console.log(`Catched error: ${error.message}`);
  } finally {
    console.log("Tests finished!");
    monitorBuffer.close();
  }
};

doTests();
```

You can compile it with: "tsc --strict -t ES2019 file_above.ts"

LudwigThePig, 2020-04-07:
This is awesome. Thank you!

closed 2020-06-13 by stale[bot]