dotintent/react-native-ble-plx #599: Monitoring Characateristcs

keyz23, 2020-01-10:
Hi everyone, I'm using this library to monitoring Characateristcs from two different devices. 
In particular I'm monitoring for 4 Characateristcs at devices (So I have 8 elements to monitoring). 

I have tried to print the data monitored and I don't have the same numbers of data for all Characateristcs. I have notice that sometimes I read the same Characateristc more time compared to the others. How can I fix this problem?

```
async notifications() {
    this.battery(this.device1, this.device2)
    this.devicesMonitoring(this.device1, this.device2)
  }

  devicesMonitoring(device1,device2) {
    
    const service = this.serviceGeneral();

    //TimeStamp 
    this.monitorTimeStamp = device2.monitorCharacteristicForService(service, this.TimeStamp, (error, characteristic) => {
      if (error) {
        this.error(error.message);
        return;
      }
      const buf = Buffer.from(characteristic.value, "base64");
      this.timeS = [0].map(index => buf.readInt16LE(index))
    })

    
    this.monitorRightAccelerazione = device1.monitorCharacteristicForService(service, this.Accelerometro, (error, characteristic) => {
      if (error) {
        this.error(error.message);
        return;
      }
      const buf = Buffer.from(characteristic.value, "base64");

      const [...acc_dx] = [0, 2, 4].map(index => buf.readInt16LE(index));
      this.arrayAccelerazioneDestro.push([this.timeS, ...acc_dx].join(":"))
    })

    
    this.monitorRightGiroscopio = device1.monitorCharacteristicForService(service, this.Giroscopio, (error, characteristic) => {
      if (error) {
        this.error(error.message);
        return;
      }
      const buf = Buffer.from(characteristic.value, "base64");
      const [...gyr_dx] = [0, 2, 4].map(index => buf.readInt16LE(index));
      this.arrayGiroscopioDestro.push([this.timeS, ...gyr_dx].join(":"))
    })

    
    this.monitorRightMagnetometro = device1.monitorCharacteristicForService(service, this.Magnetometro, (error, characteristic) => {
      if (error) {
        this.error(error.message);
        return;
      }
      const buf = Buffer.from(characteristic.value, "base64");
      const [...mg_dx] = [0, 2, 4].map(index => buf.readInt16LE(index));
      this.arrayMagnetometroDestro.push([this.timeS, ...mg_dx].join(":"))
    })

//......
```

keyz23, 2020-01-15:
Nobody can help me?

Cierpliwy, 2020-01-31:
Can you send me a logs from the native side? Did you check if values are actually sent with the same interval? What's the expected interval?

keyz23, 2020-02-05:
Yes they are send with the same interval, every 20ms.

Cierpliwy, 2020-02-07:
As I mentioned, I need more logs to help you.

closed 2020-04-14 by stale[bot]