dotintent/react-native-ble-plx #600: Question - reading battery levels

OisinOKeeffe, 2020-01-10:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [✓] I am running the latest version
- [✓] I checked the documentation and found no answer
- [✓] I checked to make sure that this issue has not already been filed
- [✓] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

So I am able to find find the devices I am looking for, I am able to list the services and characteristics, however, I do not see the battery service (0000180F-0000-1000-8000-00805f9b34fb) in the list of services returned. Should I expect to see it in the list?

I've tried:
`this.manager.monitorCharacteristicForDevice(
                    device.id,
                    '0000180F-0000-1000-8000-00805f9b34fb',
                    '00002a19-0000-1000-8000-00805f9b34fb',
                    (err, characterisitic) => {
                      console.log('this.manager.monitorCharacterstic');
                      console.log(
                        'monitor char - ',
                        device,
                        characterisitic,
                        JSON.stringify(err),
                      );
                    },
                    'monitor_battery',
                  );`

but I get:
`Service 0000180F-0000-1000-8000-00805f9b34fb for device ? not found`

Can anyone help me get a read on the battery level of my device?

keyz23, 2020-01-15:
Have you tried with the https://github.com/Polidea/react-native-ble-plx/wiki/Characteristic-Reading ? In my case the battery level is read-only :)

Cierpliwy, 2020-01-31:
Please make sure that services and characteristics are discovered and "Battery Level" exists. You can use `discoverAllServicesAndCharacteristics` and `characteristicsForService(serviceUUID: UUID)` to do it. Please check if characteristic instance contains property `isNotifiable` set to true. You can find more info about the calls in the docs.

closed 2020-04-07 by stale[bot]

Bikash-Sawa, 2021-03-07:
> Please make sure that services and characteristics are discovered and "Battery Level" exists. You can use `discoverAllServicesAndCharacteristics` and `characteristicsForService(serviceUUID: UUID)` [...]

you have example to achieve the battery percentage.