dotintent/react-native-ble-plx #601: IOS strange and random disconnects (Code=6 "The connection has timed out unexpectedly.")

SlavaInstinctools, 2020-01-17:
I'm not sure, that is a bug in this library, but I would like to find any information or opinion, that can help!

## Prerequisites

I connect a peripheral device to a phone (ios/android) and then get a large amount of data from the device using notifications. (Stream this data by subscribing on one of the characteristics - device.monitorCharacteristicForService). As the device, I use an emulator on "bleno" library.

- [+] I am running the latest version
- [+] I checked the documentation and found no answer
- [+/-] I checked to make sure that this issue has not already been filed

## Expected Behavior

I need no automatic and unexpected disconnects on IOS if the connection is good (signal and so on). As currently in android - everything is fine. 

## Current Behavior

Stably disconnections on IOS in 1-2 minutes after connection. Connection time until disconnect is random but as rule max is 2 minutes.
On android everything is okay and the connection is not interrupted and there are not any disconnects.	

By the way, I saw similar behavior with my Mi Band 4 and the official app on IOS. On Android, everything is okay too.

### Steps to Reproduce

There are two situations:

1) Just connect a peripheral device to the IOS device (bleManager.connectToDevice, bleManager.discoverAllServicesAndCharacteristicsForDevice) and wait. In 1-2 minutes device will be disconnected automatically (device.onDisconnected listener get device object and error = null) with next logs:

```
[RxBLEKit|DEBG|07:55:53.167]: CentralManager(10804420368) didDisconnect(from: Peripheral(uuid: A3F1E5A5-8D4B-C689-F87E-D706FB898478, name: Optional("my-mac")),
error: Optional(Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}))
```

2) Connect to a device, subscribe to notifications and start streaming data from device to phone. Then in 1-2 will be a disconnect. For example, here I managed to get 712 notifications and then there is an unexpected disconnect (device.onDisconnected listener get device object and error = null):

```
[RxBLEKit|DEBG|08:07:37.102]: Peripheral(uuid: A3F1E5A5-8D4B-C689-F87E-D706FB898478, name: Optional(“my-mac")) didUpdateValueFor(for:Characteristic(uuid: 6CA841C2-CE94-443C-BF2E-5F0790402164, id: 10739225088),
value: Optional("fc8481fc7ff6fc7ec2fc7f37fc8004fc84c9"),
error: nil)
[RxBLEKit|DEBG|08:07:37.103]: Peripheral(uuid: A3F1E5A5-8D4B-C689-F87E-D706FB898478, name: Optional("my-mac")) didUpdateValueFor(for:Characteristic(uuid: 6CA841C2-CE94-443C-BF2E-5F0790402164, id: 10739225088),
value: Optional("fc88effc8e2dfc8b5efc83dbfc80f9fc7c93"),
error: nil)
[RxBLEKit|DEBG|08:07:37.825]: CentralManager(10752780240) didDisconnect(from: Peripheral(uuid: A3F1E5A5-8D4B-C689-F87E-D706FB898478, name: Optional("my-mac")),
error: Optional(Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}))
Unhandled error happened: Connection error has occured: -
 subscription called from:
```

Sometimes I can get 2500 notifications without disconnects, but it's a really rare situation...

### Context

I tried finding something on the internet by this error code and message and found next:

1) https://stackoverflow.com/questions/32588325/ios-bluetooth-le-code-6-the-connection-has-timed-out-unexpectedly
2) https://devzone.nordicsemi.com/f/nordic-q-a/9224/ios-8-disconnection-code-6-the-connection-has-timed-out-unexpectedly
3) https://github.com/innoveit/react-native-ble-manager/issues/555

Looks like it's some problem or feature of apple devices, but I believe that there is a solution because it's really strange. I'm not so good at the BLE topic and probably missed some important detail and as a result, have this situation with disconnects on IOS.

But I believe that there is someone who knows what to do or had this situation too! And wait for your help or opinion :)

Cierpliwy, 2020-01-31:
Unfortunately, the best way to solve that issue is to sniff actual transmission and check if connection parameters are violated. There is a guide how to do it here: https://www.polidea.com/blog/bluetooth-low-energy-sniffing-guide/.

closed 2020-04-07 by stale[bot]

rhakh, 2020-04-26:
Found solution for me.
Last message:
https://lists.apple.com/archives/bluetooth-dev/2013/Apr/msg00154.html

ekrivenja-sequenex, 2021-11-29:
@rhakh , do you mean everything is about 2.4GHz interference?

rhakh, 2021-11-29:
@ekrivenja-sequenex  Yes. BLE and Wi-Fi (2.4 ghz) works on the same frequency. Try to find “clear” space and try again to connect devices.

sergey-g-s, 2024-07-10:
Hi @SlavaInstinctools, I am experiencing the same issue with random Bluetooth disconnections on iOS. Have you found any solution to this problem?

hardddd, 2025-03-21:
Hey Team,

I am facing same issue didDisconnectPeripheral call....Optional(Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.})

please help me i am stuck...