dotintent/react-native-ble-plx #605: Bug Report

rk85097, 2020-01-27:
## Prerequisites
- [ YES] I am running the latest version
- [YES ] I checked the documentation and found no answer
- [YES ] I checked to make sure that this issue has not already been filed - I have found another similar issue but it is not my case: https://github.com/Polidea/react-native-ble-plx/issues/346

## Expected Behavior
manufacturerData should not be null. When scanning for peripherals, then reading their manufacturerData, I expect it not to be null.

## Current Behavior
When scanning for peripherals, in case this peripheral was never scanned before, manufacturerData will be null, only when scanning again, the manufacturerData is correct. Seems that it returns null to peripherals that the mobile device never "met" before.

### Steps to Reproduce
1. Scan for peripherals.
2. While scanning, read manufacturerData

rk85097, 2020-01-27:
**UPDATE**: I have found that when `allowDuplicates` `scanOption` is `true`, `startDeviceScan` will return multiple peripherals records of the same pheripheral and sometimes the first record will have `manufacturerData` null, which means ble-plx returns a peripheral on `scanOption` before getting initial scan response from peripheral causing `manufacturerData` to be empty. This is a bug IMO.

rk85097, 2020-01-27:
**TEMP-SOLUTION** - Always take last peripheral record coming from startDeviceScan to insure manufacturer data is not empty.

Cierpliwy, 2020-01-31:
This can happen. Very often advertisement data is split into two packages and OS may report both of them after another without merging them. I'm talking about `ADV_IND` and `SCAN_RSP`. What you can do on firmware side is to make sure that manufacturer data is moved to the first packet `ADV_IND` and move less important advertisement data to the second one.

Another option is to wait for last record as you mentioned. It should have merged properties.

closed 2020-04-07 by stale[bot]