dotintent/react-native-ble-plx #607: Bug Report

ApurvMahesh, 2020-01-30:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ yes ] I am running the latest version
- [ yes ] I checked the documentation and found no answer
- [ yes ] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: X.Y.Z
* Platform: Android/iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

dariuszseweryn, 2020-01-30:
I believe you have not read [wiki regarding service discovery](https://github.com/Polidea/react-native-ble-plx/wiki/Device-Service-Discovery)

ApurvMahesh, 2020-01-30:
> I believe you have not read [wiki regarding service discovery](https://github.com/Polidea/react-native-ble-plx/wiki/Device-Service-Discovery)

May be i am not getting what you want to convey, but i am using the same method as it is described - device.discoverAllServicesAndCharacteristics(): Promise<Device>

Kindly let me know, your views on this.

dariuszseweryn, 2020-01-30:
This part:

> When the service discovery was already performed after the connection has been established it is possible to retrieve the cached services by calling:
> ```
> device.services(): Promise<Service[]>
> ```
> or
> ```
> bleManager.servicesForDevice(deviceIdentifier: DeviceId): Promise<Array<Service>>
> ```

ApurvMahesh, 2020-01-30:
> This part:

Hi @dariuszseweryn I checked the api doc and what you say is correct, but still i am having issue in discovering any services or characcteristics for the BLE device. 

Below am posting the screen shot of the response which i got from 

> device.services()

Kindly refer to the screen shot for more information.
[image]

eliasws, 2020-01-30:
I have the same Problem. Even on scanning more information is shown:

Scanning:

[image]

After discovering (same for after connecting): 

[image]

dariuszseweryn, 2020-01-30:
@ApurvMahesh You may be facing an Android OS bug but I cannot really tell without native logs as per bug report template
@eliasws these are correct as most of the data is available only during a scan

ApurvMahesh, 2020-01-31:
> @ApurvMahesh You may be facing an Android OS bug but I cannot really tell without native logs as per bug report template
> @eliasws these are correct as most of the data is available only during a scan

@dariuszseweryn I am facing the same issue as eliasws posted and it's not an Android specific issue as the screenshots i posted are related to iOS device.

Ajaykumar02, 2020-01-31:
@dariuszseweryn, I am also facing the same issue.
when tested on NRF connect application i am able list down the characteristics and services but the same thing will return null response for service[]. same on both Android as well in IOS.

pls can you look into this issue

Cierpliwy, 2020-01-31:
Ok. Let me clarify. Device has multiple properties:
```
▸ id
▸ name
▸ rssi
▸ mtu
▸ manufacturerData
▸ serviceData
▸ serviceUUIDs
▸ localName
▸ txPowerLevel
▸ solicitedServiceUUIDs
▸ isConnectable
▸ overflowServiceUUIDs
```
We can group them into explicit categories:
Read-only (stable) values:
```
▸ id
```
Properties, which can be modified, when the new Device object is fetched from the API (Device by itself is immutable and "snapshot" of it at the moment of the call):
```
▸ name
▸ mtu
```
Properties, which are fetched during the scanning procedure from the advertisement data (they are always null outside device scanning):
```
▸ rssi
▸ manufacturerData
▸ serviceData
▸ serviceUUIDs
▸ localName
▸ txPowerLevel
▸ solicitedServiceUUIDs
▸ isConnectable
▸ overflowServiceUUIDs
```
Therefore if you want to get services of the device (not services advertised during scanning procedure) you have to call sequentially `device.iscoverAllServicesAndCharacteristics()` and then `device.services()`. The result of the last call should return `Array<Service>`. I guess that's what you expect.

Why all of the properties are in one bag? That's just a mistake during early development, which we left as it is to keep backward compatibility (it's so important for me in this crazy JS world).

ApurvMahesh, 2020-01-31:
@Cierpliwy Thanks a lot for your response. I tried the same as you suggested. It is working and i am able to get the list of services and characteristics. Thanks a lot. Hereby closing the issue, as it was invalid. @dariuszseweryn thanks a lot for your help bro.

closed 2020-01-31 by ApurvMahesh

Ajaykumar02, 2020-02-01:
@Cierpliwy, cool i have not used the device.services(). Thank you for clarification.